gopro/OpenGoPro #10: Send commands to initiate live stream

jasperpants, 2021-06-23:
Would it be possible to implement commands to have the camera connect to wifi and begin a livestream via RTMP, Facebook etc?

hpwd, 2021-06-29:
I need this so much

KonradIT, 2021-06-30:
To shine some light:

I think this is done first by issuing the WiFi creds over to char `0x39` (settings). They basically do the inverse of https://gopro.github.io/OpenGoPro/ble#get-hardware-info for sending the data.

[0x39 20 28...]

[0x39 80 34...]

[0x39 81 34...]

Then the actual stream gets sent over 0x2F 

[0x2F 20 48...]

[0x2F 80 61....]

h0peixoto, 2021-07-03:
> To shine some light:
>
> I think this is done first by issuing the WiFi creds over to char `0x39` (settings). They basically do the inverse of https://gopro.github.io/OpenGoPro/ble#get-hardware-info [...]

Have you tried this bro ?

zsinba, 2021-10-17:
I need this so much

tcamise-gpsw, 2022-09-15:
Hi. This is now possible and defined in the BLE [spec](https://gopro.github.io/OpenGoPro/ble_2_0#live-streaming)

closed 2022-09-15 by tcamise-gpsw