gopro/OpenGoPro #13: Is there a BLE query to get the last file name?

dorealex, 2021-06-30:
As title indicates, is there a bluetooth query to get the name of the last file saved from either video or picture?

I am trying to control a GoPro using a raspi and I think it would be easier(and possibly better on the battery) if I just used BLE rather than WIFI.

KonradIT, 2021-06-30:
Even if you could get the last file name, how would you transmit the file data over to your RPI without using WiFi?

You can only use BLE to do basic camera controls, get the status and change settings. All things related to media or streaming are done over WiFi.

There's a third option, which GoPro hasn't *officially* documented yet, which involves using the APIs exposed over USB Ethernet to get media off the camera and even do some basic camera control (read: take pictures). Check out: https://github.com/KonradIT/goprowifihack/blob/master/HERO9/GoPro-Connect.md

dorealex, 2021-06-30:
I don't actually need the file, yet.

Basically, I have a raspi ask the gopro(which is under water) to take a picture every 5 seconds, and I have a GPS module get the location. I then build a csv file with the time, the file name, and the position. Once the "run" is done, I dump all the pictures into a folder, and run a separate script, to run the pictures through a tensorflow model, add the result of that to the CSV file and generate an interactive map. 

(also thanks Konrad, big fan of your stuff!)

Eventually, I probably would like to process the files on the raspi real-time(probably using tensorflow lite), while on the water... Which would be awesome, because my long range comms is very low bandwidth, so I probably could benefit from just transmitting the result of the model rather than transmitting the picture and processing it...

KonradIT, 2021-06-30:
Ah, interesting. Looking at the chars for data, there seems to just be camera name, battery life, wifi ssid/pass... but not last media name. The status response (https://gopro.github.io/OpenGoPro/ble#status-codes) doesn't carry it either.

dorealex, 2021-06-30:
Well, I guess I will go back to the wifi method, I just find it chews up the battery quite a bit. I also have issues with interface priority, but that's a separate issue.

leeprevost, 2021-07-12:
> Basically, I have a raspi ask the gopro(which is under water) to take a picture every 5 seconds, and I have a GPS module get the location. I then build a csv file with the time, the file name, and [...]

@dorealex : Would love to collaborate -- I have a similar application (underwater dredge -fishing application).

tcamise-gpsw, 2021-07-22:
Hello all. No, there is currently no way to explicitly get the last file name. Your best bet currently is to continuously poll the media list (only possible via WiFi). 

I can see how getting the last filename via BLE would be useful for you use case here.  I'm not sure it makes sense to expose the entire media list functionality over BLE due to bandwidth problems as the media list can get quite large but maybe a subset like "last media captured" would make sense.

I'll mark this as an enhancement and leave it open.

dorealex, 2021-07-27:
> @dorealex : Would love to collaborate -- I have a similar application (underwater dredge -fishing application).

@leeprevost  I've created a new repository with my stuff. Bear with me, I don't know exactly what I am doing. https://github.com/dorealex/snoop

leeprevost, 2021-07-27:
Thank you.  I’ll check it out.

lozo2010, 2021-11-29:
Hello. I'm very interested in this functionality. Is there an expected date?

EddyCelis, 2022-04-23:
Add +1 for enhancement request. I'm recording 40 sec clips and storing related data in a database. It would be great if I could write the filename to the database immediately instead of trying to figure it out later (Wifi file list delta and sequence number of recording in database).

EddyCelis, 2022-09-15:
> @lozo2010 @EddyCelis @dorealex I am starting to look into this. Is it
> acceptable for your use cases for this command to only be exposed over
> Wifi? Or do you need it for BLE also?

That would be BLE only.
On WiFi it is already possible to handle that in a roundabout way by
getting the file names before recording starts and then after recording
stopped, the new file would be... well, the new one. But that would take
too long.
What we want to do is;
- start recording
- stop recording
- get a characteristic push with the filename that was just recorded.

We want to add this filename into our SQLite database together with the
other info that is specific to our app that is linked to the video.

Thanks,
Eddy

tcamise-gpsw, 2022-09-15:
It is very unlikely that you will get the push functionality. But I think it is feasible to add a command to poll via BLE (and / or Wifi) for the last filename. I'll take a look

dorealex, 2022-09-15:
Eddy's right, for my use case its a picture, but I guess its all the same. Could the filename be stored as variable similar to the SSID whenever it is stored to the SD card,  and then made accessible similarly via BLE.

tcamise-gpsw, 2022-09-15:
We can't add more characteristics. I'm proposing a new [command](https://gopro.github.io/OpenGoPro/ble_2_0#commands).

So if you're doing it programmatically (i.e. capturing the media by sending an OGP command), it's pretty straightforward:
- Set Shutter
- GetLastFilename (new command)

I guess the problem is that if you also need the last filename for media captured by the physical shutter button. You can already accomplish by registering for status updates for `num total videos` or maybe even `remaining space` so that you know when new media is captured. So:
- register for status update
- wait for status update
- GetLastFileName

Does that work?

EddyCelis, 2022-09-15:
A command to get the last filename via BLE would be fine.

Eddy

lozo2010, 2022-09-16:
My use case is the same as @EddyCelis. In my case, they never use the camera button to record.
With the command to get the last name, after stopping the recording it would be perfect

EddyCelis, 2022-09-19:
Hi Tim,

 

To be sure I was never thinking of adding a complete new characteristic.

 

I was thinking of adding a status ID that would then be reported like the other status IDs. Either through a direct command (query id 0x13) or as part of a status update that was registered with the ‘register for status value updates’ (query Id 0x53). These go out on GP-0076 and the responses come on GP-077. That way you could get the last filename unsolicited (through the status update) or directly by sending a status query with the new query ID, almost the same as a dedicated command but using the get status command. Status ID is a full byte, so it looks like there is some room to play still. If it is 8 bits in the firmware as well.

 

But a dedicated command will also work.

 

For a burst it is probably best to get the name of the first picture in the sequence and not the last.

 

Best regards,

Eddy

tcamise-gpsw, 2022-09-28:
> My use case is the same as @EddyCelis. In my case, they never use the camera button to record. With the command to get the last name, after stopping the recording it would be perfect
> 
> @tcamise-gpsw Would this command be possible for the Hero7?

Definitely not for hero 7. Any changes need to be incorporated into future firmware builds, at which time are currently only planned for Hero 11.

tcamise-gpsw, 2022-09-28:
> Hello. I'm very interested in this functionality. Is there an expected date?

I can't provide a good estimate right besides hopefully end of the year.

EddyCelis, 2022-09-28:
That is disappointing.

If you look at the functionality of the interfaces from Hero 8 on up, any limitations in the interface between the versions was always due to hardware constraints, never pure firmware. Is there a reason for not supporting it on the Hero 8 – 10?

 

This limits the target market in a big way if the app is relying on this functionality. It would pretty much be “Hero 11 only”.

EddyCelis, 2022-10-11:
We are monitoring the 'encoding' state, so setting shutter from our app or
the physical button pushes our state machine to the same state, so that
does not make a difference.

- send 'shutter off'
- 'encoding' state goes False
- EITHER filename state push is received when all the files are ready.
Probably on characteristic 77?
- OR send new command 'GetLastFileName(s)'

Note that I did add the (s) to filename, because I believe that in burst
mode, the 'Encoding' state stays True for the entire burst?
So that could mean lot of files.

In our application we do not use burst mode, so not important for us. But
if you implement it, you might have to return all the file names for the
last burst. Or document that it will only return the last one.

Eddy

EddyCelis, 2023-02-24:
Has there been any decision on this enhancement suggestion?

tcamise-gpsw, 2023-12-06:
FYI this will be released in the next firmware update, before the end of the year,

tcamise-gpsw, 2023-12-07:
It will be Hero12 and newer.

lozo2010, 2023-12-10:
This improvement was needed on cameras that didn't have NCM, as the Wi-Fi
connection was unstable. From the gopro 10 onwards I have no need for this.
It's a shame it's not for previous cameras. I'll have to keep fighting with
the wifi of my 15 gopro hero 7.
Thank you anyway for your work on this

EddyCelis, 2023-12-11:
It will allow us to add an important feature to our app, unfortunately only for the new Heros and the ones in the future.
Better than nothing. Thanks for the update!

tcamise-gpsw, 2024-04-10:
I forgot to update this ticket but this functionality has been added several releases ago. Documented in [BLE ](https://gopro.github.io/OpenGoPro/ble/features/query.html#get-last-captured-media) and [HTTP](https://gopro.github.io/OpenGoPro/http#tag/Media/operation/OGP_GET_LAST_MEDIA)

closed 2024-04-10 by tcamise-gpsw