gopro/OpenGoPro #14: Error on running gopro-video demo

leeprevost, 2021-07-03:
Got error messsage when running demo.

$ gopro-video
Traceback (most recent call last):
  File "c:\users\lee\anaconda3\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\lee\anaconda3\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\lee\Anaconda3\Scripts\gopro-video.exe\__main__.py", line 4, in <module>
  File "c:\users\lee\anaconda3\lib\site-packages\open_gopro\__init__.py", line 7, in <module>
    from open_gopro.gopro import GoPro
  File "c:\users\lee\anaconda3\lib\site-packages\open_gopro\gopro.py", line 28, in <module>
    from open_gopro.ble_commands import BleSettings, BleStatuses, BleCommands, BLECommunicator
  File "c:\users\lee\anaconda3\lib\site-packages\open_gopro\ble_commands.py", line 269
    def set_shutter(self, shutter: params.Shutter, /) -> GoProResp:
                                                   ^
SyntaxError: invalid syntax

daverusso, 2021-07-04:
Just guessing .... but the top-level [SDK requirements](https://github.com/gopro/OpenGoPro#requirements) seem to imply that the demos don't run natively on Windows; you must run from within a Docker image, which has all the right versions of python, all tools, packages, etc.

leeprevost, 2021-07-05:
After inspecting code, I think it would likely run if I pass the id, and paths arguments via cli.  I’ll try it and comment back.

leeprevost, 2021-07-06:
A little more on this problem.   I think this may be due to a python version conflict I have on my machine.   I assume the gopro-stream.exe runs "python" on my machine and passes the demo module to it.   My python path maps to version 3.7 branch.   But, my py path maps to a 3.8 version.    I think I've seen somewhere that python 3.8 is required?   am trying to upgrade my environmnt to see if this is the problem.

KonradIT, 2021-07-06:
I think the position only parameter symbol (,/) is to blame, which is used in python3.8: https://www.python.org/dev/peps/pep-0570/ 

So try with python3.8 .

daverusso, 2021-07-06:
I didn't see 3.8 mentioned in the docs, but I noticed it in the open-gopro package's `PKG-INFO`:

```
Classifier: Programming Language :: Python :: 3
Classifier: Programming Language :: Python :: 3.8
```

leeprevost, 2021-07-06:
I did try by activating a virtual environment with version 3.8 in it.   I'm not how gopro-stream.exe calls python but if its using my virtual enviromnt python where I executed it from, I still get the error.

(stockdata) PS C:\Users\lee\Dropbox\python sandbox\OpenGoPro> python --version
Python 3.8.10
(stockdata) PS C:\Users\lee\Dropbox\python sandbox\OpenGoPro> gopro-stream
Traceback (most recent call last):
  File "c:\users\lee\anaconda3\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\lee\anaconda3\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\lee\Anaconda3\Scripts\gopro-stream.exe\__main__.py", line 4, in <module>
  File "c:\users\lee\anaconda3\lib\site-packages\open_gopro\__init__.py", line 7, in <module>
    from open_gopro.gopro import GoPro
  File "c:\users\lee\anaconda3\lib\site-packages\open_gopro\gopro.py", line 28, in <module>
    from open_gopro.ble_commands import BleSettings, BleStatuses, BleCommands, BLECommunicator
  File "c:\users\lee\anaconda3\lib\site-packages\open_gopro\ble_commands.py", line 269
    def set_shutter(self, shutter: params.Shutter, /) -> GoProResp:
                                                   ^
SyntaxError: invalid syntax

leeprevost, 2021-07-06:
After realizing I was probalby running into a python version issue, I followed @daverusso suggestion to setup a virtual environmnt with python=3.8.    I reported above what happened when I tried to run in the env.  But, after doing the pip install of open-gopro within my env, I'm now gettiing a different error:

`(gopro) PS C:\Users\lee\anaconda3\envs\gopro\scripts> gopro-demo
ERROR    19:01:05.888 ValueError("invalid literal for int() with base 10: 'file'")                                                                                                                                               demo.py:182
Exiting...
(gopro) PS C:\Users\lee\anaconda3\envs\gopro\scripts> gopro-stream
⠋ Connecting...ERROR    19:01:30.471 ValueError("invalid literal for int() with base 10: 'file'")                                                                                                                                              stream.py:56
Exiting...
⠋ Connecting...`

KonradIT, 2021-07-06:
Ran into this issue (https://github.com/gopro/OpenGoPro/pull/8) as well, make sure you run the latest opengopro python code (try cloning it instead of installing via pip)

leeprevost, 2021-07-07:
Thanks @KonradIT .  Success!

Here was my process:
1) Create virtual env per @daverusso suggestion.
2) I used pip to install open-gopro.  I uninstalled.
3) I then cloned source into my project directory.
4) I pathed into the python subdirectories that had setup.py and ran the python setup

`Thanks @KonradIT .  Success!

Here was my process:
1) Create virtual env per @daverusso suggestion.
2) I used pip to install open-gopro.  I uninstalled.
3) I then cloned source into my project directory.
4) I pathed into the python subdirectory (sdk_wireless ....)  that had setup.py and ran the python setup

`(gopro) PS C:\Users\lee\anaconda3\envs\gopro\scripts\src\open-gopro\demos\python\sdk_wireless_camera_control> python setup.py install`

Victory!`

Victory!

KonradIT, 2021-07-07:
Hmm would be nice if GoPro released the fix currently in main branch to pypi

tcamise-gpsw, 2021-07-07:
Hi all. Just to clarify, you should be able to do the following:
1. Set up a new virtual environment using Python 3.8.x
2. `pip install open-gopro`
3. 'gopro-video`

Some of the above errors look like conflicting python packages / versions which is not unexpected if you are not using virtual environments or if you are using a different Python version than 3.8

If you are running into something else after doing the above steps, let me know and I'll try to reproduce.

FYI this is very much still in alpha / beta but we are actively working on improving it.

Also, there is some future work to support other Python versions besides 3.8.x

tcamise-gpsw, 2021-07-07:
> Hmm would be nice if GoPro released the fix currently in main branch to pypi

@KonradIT I will work on doing that this week.

KonradIT, 2021-07-07:
Note that on Windows, @leeprovost will need this PR (https://github.com/gopro/OpenGoPro/pull/11) to be on the code. The latest PyPi release is dated May 27th. So the bug fix is not there. 

@tcamise-gpsw I actually had this issue and could reproduce it, and made a PR. Just in case, would it be possible to have an updated Python package on PyPi based off the latest `main` code? Thanks.

tcamise-gpsw, 2021-07-07:
Yep, you're correct. Sorry for the confusion.

Anyway, I just published the fix at version [0.5.7](https://pypi.org/project/open-gopro/)

leeprevost, 2021-07-08:
@tcamise-gpsw or @KonradIT - thanks for your comments on this.   

I wonder if you could direct me to a discussion area where I could post a job spec for hiring a project devloper for a small opengopro project.    My target system is problably linux/pizero but likely develop on windows.  mainly looking for initialization/pairing scripts, sleep/wake, and video stream on scripts.   Likely embedded with web server and flask.   

Also, i'm not a hardcore developer and I must share that I've had a pretty rough time wth the windows user experience with this.  I've gotteen through the tutorials and the demos but with the problems you see above.  I'm also running into what I think are problems with how Blink handles cross platform.  it seems like the process for scanning, pairing, and then connecting are different for windows vs. linux?     I have trouble getting gopro to reconnect after it goes to sleep and think the standard demo scripts try to repair the device.     You can see my experince here:  https://github.com/hbldh/bleak/issues/367#issuecomment-875983813 posted on the bleak issues list (but which may be more relevant here).   Apprciate any help you can give me here.

daverusso, 2021-07-08:
FWIW: after a `pip install open-gopro --upgrade` to get `0.5.7`, I was able to connect, record a 10 video, and download it on Windows 10 (woohoo!).  

Thanks to both @KonradIT and @tcamise-gpsw!  Really nice to see such rapid support.

I struggled a bit this morning due to bonehead mistakes on my side: 

  - my phone was paired with and connected to my freshly unboxed hero9 (doh!)  
  - I had connected the hero9's USB to my desktop's USB to charge the battery (and keep its bluetooth stack running, I think).  But during the connection process, the hero went into a mode that allows Windows to access the hero's SD card(?).  Switching to a wallwart USB adapter avoids this issue (obviously).

That said, I still have to run the `gopro-video` multiple times before it's able to get through the connect, record, and download without failing at one of these steps.  Still trying to figure out how best to setup the camera to make this robust.   But not bad progress for the first 1/2 day.

```
(gopro) c:\Users\dr\OpenGoPro-main\demos\python\sdk_wireless_camera_control>gopro-video 10
⠼ Connecting...WARNING  12:01:31.156 Failed to find a device in 5 seconds. Retrying #1                                    gopro.py:381
⠼ Connecting...WARNING  12:01:59.129 BleakError('Could not start notify on 00002a19-0000-1000-8000-00805f9b34fb:          gopro.py:408
         Unreachable'). Retrying #1
⠴ Connecting...WARNING  12:02:00.853 BleakDotNetTaskError('Could not get GATT characteristics for                         gopro.py:408
         Windows.Devices.Bluetooth.GenericAttributeProfile.GattDeviceService: AccessDenied'). Retrying #2
⠴WARNING   Connecting...12:02:15.943 TimeoutError(). Retrying #3                                                          gopro.py:408
⠸ Connecting...WARNING  12:02:31.054 TimeoutError(). Retrying #4                                                          gopro.py:408
ERROR    12:02:31.054 ConnectFailed('GoPro Error: BLE connection failed to establish after 5 retries with   video.py:59
         timeout 15')
Exiting...

(gopro) c:\Users\dr\OpenGoPro-main\demos\python\sdk_wireless_camera_control>gopro-video 10
         12:02:57⠦. 437Connecting... Unhandled notification from handle 15                                                gopro.py:424
Success!! 😃 File has been downloaded to video.mp4
Exiting...
```

Finally, thanks for the very nice `video.log` output file!  I still have to come up to speed on bluetooth, but this file looks like there's plenty of good debug info.  It's time to dig into the code.

leeprevost, 2021-07-09:
Looks like awesome progress @daverusso.  That process looks much smoother!   Nice improvement.   

I too am getting inconsistent results with the demos and tutorials particulary after GoPro sleeps.   I wonder if the routine needs to:

1. Scan/discover
2. Pair if previously unpaired.   (my review of the first tutorial and its subsequent use in other tutorials it always pairs, even if previously paired).   
3. Connect
4. control

I referred to my issues post on the bleak repo above https://github.com/hbldh/bleak/issues/367#issuecomment-875983813 to what I'm finding.

daverusso, 2021-07-09:
> I too am getting inconsistent results with the demos and tutorials particularly after GoPro sleeps.
Yep, keeping the USB plugged in keeps the bluetooth discoverable even after a "power down".

I also suspect Windows is partly to blame, I really should be using the Raspberry or even WSL2.  I'm asking for trouble by using my desktop run the demos: when downloading the video, for example, Windows must disconnect from my LAN and connect to the hero's access point.  This no doubt takes much longer than the demos expect and triggers events that would unnecessarily complicate demos that really should be running on an embedded device.  

That said, I've done things like this in the past, so I'm optimistic that I can make progress creating a development environment before moving the code to the Raspberry.

leeprevost, 2021-07-09:
> I also suspect Windows is partly to blame, I really should be using the Raspberry or even WSL2

Me too.  

But that is what I thought bleak was supposed to do for us - ie make Bluetooth platform neutral/independent.  My eventual target application is rpizero but it seems many would experience this inconsistency even as developers while trying first experience in windows.

KonradIT, 2021-07-09:
There isn't really an alternative to `bleak`, on my unofficial GoPro BLE program I used pygatt which only works on Linux or other OSes that can run BlueZ and it somehow ran even worse than bleak. I now run a Windows machine and haven't had any issues with bleak. bleak also runs well on raspbian.

What was your pairing procedure with Windows?

leeprevost, 2021-07-10:
It was using the connect_ble.py in the tutorials.  Also, you can see my link to the issue log on bleaks repo above.

tcamise-gpsw, 2021-07-22:
Yep, I unfortunately see similar issues to what you are running into sometimes. They do appear to be bleak / OS based.  

I agree that there needs to be more work done here to make everything more robust via both digging into the issues to file bleak bugs as needed as well as handling errors better in Open GoPro.

Ultimately though, I have a suspicion that the Windows bluetooth drivers are just not very stable. In any case, I'll close this since the original issue is fixed.

closed 2021-07-22 by tcamise-gpsw

tcamise-gpsw, 2021-08-19:
@leeprevost I somehow missed your link to the bleak issue above but ended up just stumbling there independently! Anyway, I wish I would have seen that because this is certainly an issue in the Open GoPro repo. I've opened a new bug to fix it: #39

leeprevost, 2021-09-03:
> @leeprevost I somehow missed your link to the bleak issue above but ended up just stumbling there independently! Anyway, I wish I would have seen that because this is certainly an issue in the Open GoPro repo. I've opened a new bug to fix it: #39

@tcamise-gpsw Lots of new discussion going on #hbldh/bleak#367 but it seems folks are mainly focused on linux.   I think its a windows problem too.