gopro/OpenGoPro #19: GoPro class handler -- no battery status?   Doesn' turn off wifi AP?

leeprevost, 2021-07-15:
As I've climbed my learning curve spending time with both the tutorials and the demos, I'm beginning to understand just how powerful the [GoPro class handler](https://github.com/gopro/OpenGoPro/blob/b3e435ab25cbd12a9ea1db9a511e0425a9c6d402/demos/python/sdk_wireless_camera_control/open_gopro/gopro.py#L85) is for us.   It handles all the down and dirty asynchronous programming, establishes loops and threads, and does keep alive for those of us who do not fully understand asynchronous I/O programming techniques.  

That said, and it could be somthing I'm missing or misunderstand about the abstraction, I don't think it is doing what its meant to do arond the areas of disabling the WiFi Access Point and returning the battery status.   

My simple test was to establish the [GoPro ](https://github.com/gopro/OpenGoPro/blob/b3e435ab25cbd12a9ea1db9a511e0425a9c6d402/demos/python/sdk_wireless_camera_control/open_gopro/gopro.py#L85)class using the convenience method rather than the context handler as outlined in its docs.   As I note GoPro connects, establishes wifi and subscribes to notifications in its initialize and __enter__() methods, I tried to then disable wifi in order to conserve power and also monitor the battery conditions.

Here is my simple setup:

```
from open_gopro import GoPro
import logging
import time

logging.basicConfig(filename="gopro_test.log",level=logging.DEBUG)

#connect to camera, start services, etc.   Alternative to context manager usage with start() convenience method.
gopro = GoPro("2659")  #my camera identifier
gopro.start()
# if wifi ap is on, turn it off to save battery
if gopro.wifi.is_on:
    gopro.ble_command.enable_wifi_ap(False)

try:
    #see if it turned off
    assert not gopro.wifi.is_on
except Exception as e:
    #log the assertion error if it didn't
    logging.error(repr(e))

#now -- while keep alive is continously connecting, log battery level and see if it changes over time.  Look for final
#last gasp before battery level shuts down gopro.  Sample every 60seconds recording battery level.
samples = []
for x in range(0,1000):
    sample = (x, time.asctime(), gopro.ble_status.batt_level.id.value)
    samples.append(sample)
    print(sample)
    logging.info(sample)
    time.sleep(60)
```

Snippets from my log file are as follows:

- Camera scan discovers camera and connects correctly including disovery of wifi, password, SSID, etc.  As exepected, gopro.start() calls convenience method to enter the context stat using [GoPro.__enter__()](https://github.com/gopro/OpenGoPro/blob/b3e435ab25cbd12a9ea1db9a511e0425a9c6d402/demos/python/sdk_wireless_camera_control/open_gopro/gopro.py#L186)  method both runs self.initialize() and leaves the AP in "on" state.   i also see the first of the keep alive log entries, again, as expected.

- Here's where it seems to go off the rails.   I then use what would intuitively seem to be a status flag for wifi -- gopro.wifi.is_on which returns True and I then attempt to disable the wifi AP to conserve power:

```
DEBUG:open_gopro.wifi_controller:Waiting 1 second for Wi-Fi connection to establish...
DEBUG:open_gopro.util:Send cmd --> netsh wlan show interfaces
DEBUG:open_gopro.util:Receive response --> 

There is 1 interface on the system: 

    Name                   : Wi-Fi

    Description            : Killer Wireless-n/a/ac 1435 Wireless Network Adapter

    GUID                   : 77ff450e-786f-48ea-a991-82ff16779da7

    Physical address       : 9c:b6:d0:95:35:09

    State                  : connected

    SSID                   : syr h9 black dredge1

    BSSID                  : d6:32:60:ae:c8:4c

    Network type           : Infrastructure

    Radio type             : 802.11ac
[... 89 lines cut ...]
DEBUG:open_gopro.gopro:SettingId.LED released the semaphore
INFO:open_gopro.ble_commands:-----------> 
{
    "status": "SUCCESS",
    "id": "UUID.CQ_SETTINGS_RESP::SettingId.LED"
}
INFO:root:(142, 'Wed Jul 14 18:37:11 2021', 2)
INFO:open_gopro.ble_commands:<----------- set : LED.BLE_KEEP_ALIVE
DEBUG:open_gopro.gopro:SettingId.LED acquiring semaphore
DEBUG:open_gopro.gopro:SettingId.LED has semaphore
DEBUG:open_gopro.ble_controller:Writing to b5f90074-aa8d-11e3-9046-0002a5d5c51b: 03:5b:01:3d
DEBUG:bleak.backends.dotnet.client:Write Characteristic b5f90074-aa8d-11e3-9046-0002a5d5c51b : bytearray(b'\x03[\x01=')
DEBUG:bleak.backends.dotnet.client:_ConnectionStatusChanged_Handler: 0
DEBUG:open_gopro.ble_controller:On device BleakClientDotNet (F5:A2:15:A9:A4:F7), Disconnected callback called!
ERROR:open_gopro.gopro:Response timeout of 10 seconds!
INFO:open_gopro.ble_commands:<----------- set : LED.BLE_KEEP_ALIVE
DEBUG:open_gopro.gopro:SettingId.LED acquiring semaphore
INFO:root:(143, 'Wed Jul 14 18:38:11 2021', 2)
```
**### So, I see several issues/questions here:**

1. It would seem the gopro.wifi.is_on status flag does not work as it would seem as it reports True even after executing gopro.ble_command.enable_wifi_ap(False) to turn off wifi AP.
2.  gopro.ble_command.enable_wifi_ap(False) doesn't seem to disable the wifi access point as I would have expected.
3. If I'm intepreting the battery status response correctly (gopro.ble_status.batt_level.id.value), it never changes from "2" even though actual battery level goes from 90%+ to 0% over this test period and logged each minute.
4. I would have expected a "last gasp" error reported through the services that the ble client subscribed to when initializing including battery status.   I think I have seen occassional errors where this interrrupt/battery status reported 0% when iniitializing (ie when battery is full and I'm establishing intial ble connection).  However, this does not appear to be subscribed corrrctly at the end.   My final error is noted in the log above (ERROR:open_gopro.gopro:Response timeout of 10 seconds!) before my last logging of battery condition which is still "2".

tcamise-gpsw, 2021-07-22:
I think you have certainly hit some valid bugs here. Hopefully you are enjoying being an early adopter 😄 

Just as some background, the whole WiFi driver is not extremely well tested right now beyond what is accomplished via the demos. I would have preferred to use an existing module but couldn't find anything cross-platform. So this is a more ambitious project than I would have liked and at some point will likely be separated out into it's own Python package.

Regarding the WiFi questions (1 and 2), I don't think I have tested this functionality yet so will try this out and get back to you.

Regarding the battery level, something smells wrong about your code here:
`gopro.ble_status.batt_level.id.value`

In any case, give me a day or two to set this up and I'll get back to you with some more info.

leeprevost, 2021-07-23:
I am enjoying being on the bleeding edge!   Snd yes, I guessed on where battery status was in that object.

tcamise-gpsw, 2021-07-29:
Hey sorry I haven't forgotten about this...just haven't had a chance to get to it yet.

leeprevost, 2021-07-30:
Thanks for the update.

tcamise-gpsw, 2021-07-30:
I got around to taking a look at this today. Firstly, I have a pull request (linked here) that shows the changes as well as added a "poll_battery.py" demo that accomplishes what you are trying to do.

Regarding your specific questions / code:

### Checking if the camera's WiFi is on

Your code:
```
# if wifi ap is on, turn it off to save battery
if gopro.wifi.is_on:
    gopro.ble_command.enable_wifi_ap(False)
```

The `wifi` attribute of the `gopro` GoPro instance here is actually the wifi interface of the client (i.e. your computer). So what you are actually checking here is that your computer's wifi is on. A couple notes;
- This property wasn't implemented in Windows. I added that in the PR
- I understand your confusion here. To try to clear this up, I marked the ble and wifi attributes of the gopro class as internal attributes of the gopro class (by adding underscores in front as is Python standard). I also slightly updated the argument description to state "computer's wifi connection"

To accomplish what you really want to do, you need to read the camera's AP State status: status 69 from the [BLE Spec](https://gopro.github.io/OpenGoPro/ble#status-codes).  This can be done (and you can see the full example in the new demo in the PR) as such:
```
camera_wifi_on =  gopro.ble_status.ap_state.get_value().flatten
```
I'm using the `flatten` attribute which is a convenience method to distill the returned response down to a bool. Alternatively, you could manually parse the returned event:
```
response = gopro.ble_status.ap_state.get_value()
camera_wifi_on = response.data[StatusId.AP_STATE]
```

Your code to disable the wifi does look correct.

I have found myself sometimes wanting this same functionality. That is, I sometimes would like to connect the camera only via BLE and don't need WiFi. Therefore, I added an optional argument to the `GoPro` class to allow this. So you can now do:
```
        with GoPro(identifier, enable_wifi=False) as gopro:
            # Ensure that wifi is not enabled. Not needed but left for instructive purposes
            assert not gopro.ble_status.ap_state.get_value().flatten
```

### Getting the battery level

Your code:
```
gopro.ble_status.batt_level.id.value
```
There is definitely a problem here. I'll try to state what each part of this is doing:
- gopro: the main `GoPro` interface which controls the camera
- ble_status: the grouping of ble status delegates which build and send status commands
- battery level: a `Status` instance to interact with the battery level status

That's all good so far. If we look at the documentation for [Status](https://gopro.github.io/OpenGoPro/python_sdk/api.html#open_gopro.ble_commands.Status), you can see that the `id` attribute which you are reading is a constant value. It is the ID from the [spec](https://gopro.github.io/OpenGoPro/ble#status-codes), i.e. 2 for Internal Battery Level.
What you really want to be doing here is calling the `get_value` method to send a command over the air to read this status's value, i.e.:
`gopro.ble_status.batt_level.get_value().flatten`
Same note here on `flatten` as above.

FYI I tried to add some examples for this in the [usage](https://gopro.github.io/OpenGoPro/python_sdk/usage.html#settings-and-statuses). I think its pretty good but if you feel like anything should be added, let me know.

Also note that the status you are getting is actually the "bars" that are shown on the camera, i.e. status ID 2 from the [spec](https://gopro.github.io/OpenGoPro/ble#status-codes). If you want the percentage you should use 
`gopro.ble_status.int_batt_per`

### Pull Request

I'll add you to the PR if you want to review it. I still need to do some cleanup to get it to pass documentation / lint tests.
If you have any ideas for future improvements or questions for clarification, let me know. Thanks!

tcamise-gpsw, 2021-07-30:
hmmm I can't actually add you as a reviewer. I'll need to work with the org owner here to allow this as this is not very conducive to public contributions...

leeprevost, 2021-08-01:
Much thanks @tcamise-gpsw.  I read through your response and it all looks straighforward.   marking attributes as internal attributes with underscore sounds dead on to me.   Also, thanks for the enable_wifi argument in the class.  I think I may still be confused as to distinctictions you are making around the battery status.  I think I now know what to do to accomplish what I want but I wonder if "ble_status" is the thing that misleads me (and others) into thinking this method returns status.  It is rather a ble_controller" which as to be told to get_status?    I'd be glad to review and provide comments but I'm not worried about being an official reviewer.

leeprevost, 2021-08-02:
Couple of thoughts after spending some more time with this @tcamise-gpsw:

1. Very big improvement and advancement in my own understanding.  thank you.
2. I still have a lot of seeming instability in trying to get GoPro to connect after pairing.  after thinking about this, I think this may be becasue my ios app was also open and trying to pair with the camera while my python app on Windows was doing the same.   I think it may be important to show a scan of active devices to allow for troubleshooting connection issues or maybe a faq on that.
3. I think a nice adder would be to to save the current wifi network before switching to the GoPro wifi, and then putting that back in place when the context manager closes.   thats a nice to have as I find it leaves me connected to my gopro on exit.
4. I still wonder why the notification services including the battery level doesn't provide some "last gasp" alert before the battery is exhausted.  That doesn't seem to happen.
5. After your enhancements and reworking my battery status understadning (ie. using gopro.ble_status.int_batt_per), I find it odd that the battery exhausts at same rate whether with AP on or not.   I did an experiment by turning AP on and then logging battery level for 10 minutes.  I then turned AP off for 10 minutes and logged level.  The experiment with AP off actually showed the rate of battery use as slighly more which is odd.  So, I do wonder what is really going on there.   I would think the AP status would make a big difference in battery usage but it doesn't seem so.  If you think any of this would be helpful, I'd be glad to help out.

tcamise-gpsw, 2021-08-02:
2. Yes, the scenario you describe can certainly lead to unexpected behavior as the GoPro only handles one connection at a time. Can you post the detailed log for the pairing fails? Also, what do you mean by "scan of active devices?" In any case, I agree we need to provide more meaningful information on failures to connect / pair and likely some documentation around this.
3. That's a good idea! I'll work on adding this to the current PR.
4. I assume you mean that you have registered for notification updates of the battery percentage (i.e. you are not polling). In this case, the notification is only sent when the battery percentage changes. So I would expect the last notification you receive to be when it hits 1%. 
5. These are good questions. My initial guess, at least based on your analysis, is that the WiFi / BLE usage is negligible to the total power consumption. But really this needs a deeper investigation as that doesn't sound right.  Perhaps we need a control test where you just leave the camera on with all wireless powered off and see how long it lasts for.  In any case, I'm currently mostly concerned with the Python SDK stability / robustness at this point, at least for this ticket.  If you want to discuss this / investigate further, can you make a separate Discussion for this?

leeprevost, 2021-08-02:
2 - would you like me to send you the log?  I think it appended multiple sessions from yesterday and is pretty long.  Lots of captured Timeout retries but also multiple attempts to connect to my local wifi.   10k lines long.
3 - great!
4 - I was thinking the notification services would alert a low battery condition but I must not understand that properly.
5 - I agree, a future wierd challenge to tackle.  Would be glad to and with  your new example battery test script, its a valid discussion.

leeprevost, 2021-08-02:
More on item 2 -- The key log errors I seem to see are:

```
Failed to find a device in 5 seconds. Retrying #1
BleakError('Could not start notify on 00002a19-0000-1000-8000-00805f9b34fb: Unreachable'). Retrying #1
BleakDotNetTaskError('Could not get GATT characteristics for Windows.Devices.Bluetooth.GenericAttributeProfile.GattDeviceService: AccessDenied'). Retrying #2
TimeoutError(). Retrying #3
TimeoutError(). Retrying #4
```
have seen this a few times:
```
Received non-success status: ErrorCode.ERROR
Received non-success status: ErrorCode.ERROR
```
Also, I get these but they seem to go away when I do give identifier to the GoPro() handler:

```
Unhandled notification from handle 15
Traceback (most recent call last):
  File "batt_test.py", line 52, in <module>
    with GoPro(args.identifier) as gopro:
  File "C:\Users\lee\Anaconda3\envs\gopro\lib\site-packages\open_gopro\gopro.py", line 203, in __enter__
    self.establish_wifi()
  File "C:\Users\lee\Anaconda3\envs\gopro\lib\site-packages\open_gopro\gopro.py", line 649, in establish_wifi
    if self.wifi.connect(self.ssid, self.password, timeout=timeout):
  File "C:\Users\lee\Anaconda3\envs\gopro\lib\site-packages\open_gopro\wifi_controller.py", line 176, in connect
    return self._driver.connect(ssid, password)
  File "C:\Users\lee\Anaconda3\envs\gopro\lib\site-packages\open_gopro\wifi_controller.py", line 901, in connect
    raise Exception("Wi-Fi connection timeout.")
Exception: Wi-Fi connection timeout.
```
Same for this one:
```
BleakError('Could not start notify on 00002a19-0000-1000-8000-00805f9b34fb: Unreachable'). Retrying #1
TimeoutError(). Retrying #2
TimeoutError(). Retrying #3
TimeoutError(). Retrying #4
Traceback (most recent call last):
  File "batt_test.py", line 52, in <module>
    with GoPro(args.identifier) as gopro:
  File "C:\Users\lee\Anaconda3\envs\gopro\lib\site-packages\open_gopro\gopro.py", line 200, in __enter__
    self.establish_ble(self._device)
  File "C:\Users\lee\Anaconda3\envs\gopro\lib\site-packages\open_gopro\gopro.py", line 409, in establish_ble
    raise ConnectFailed("BLE", timeout, retries)
open_gopro.interfaces.ConnectFailed: GoPro Error: BLE connection failed to establish after 5 retries with timeout 15
```

Also, this was an odd one -- I ran a battery test and battery level got down to 14%.   My program was going to self terminate and logs its samples but it failed with this error:

```
{'sample': 100, 'ts': 'Mon Aug  2 18:08:33 2021', 'ap_status': True, 'battery_pct': 15}
{'sample': 101, 'ts': 'Mon Aug  2 18:09:33 2021', 'ap_status': True, 'battery_pct': 14}
Response timeout of 10 seconds!
Traceback (most recent call last):
  File "C:\Users\lee\Anaconda3\envs\gopro\lib\site-packages\open_gopro\gopro.py", line 551, in write
    response = self._sync_resp_ready_q.get(timeout=WRITE_TIMEOUT)
  File "C:\Users\lee\Anaconda3\envs\gopro\lib\queue.py", line 178, in get
    raise Empty
_queue.Empty

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "batt_test.py", line 67, in <module>
    batt_stat = get_batt_status(gopro)
  File "C:\Users\lee\Dropbox\python sandbox\dredge_pro\gopro\utils.py", line 26, in get_batt_status
    return gp.ble_status.int_batt_per.get_value().flatten
  File "C:\Users\lee\Anaconda3\envs\gopro\lib\site-packages\open_gopro\ble_commands.py", line 371, in log_query
    response = wrapped(*args, **kwargs)
  File "C:\Users\lee\Anaconda3\envs\gopro\lib\site-packages\open_gopro\ble_commands.py", line 635, in get_value
[... 58 lines cut ...]
  File "batt_test.py", line 62, in <module>
    stream(gopro)
  File "C:\Users\lee\Dropbox\python sandbox\dredge_pro\gopro\utils.py", line 35, in stream
    assert gp.wifi_command.stop_preview_stream().is_ok
  File "C:\Users\lee\Anaconda3\envs\gopro\lib\site-packages\open_gopro\wifi_commands.py", line 85, in _wrapper
    response = instance.communicator.get(url)
  File "C:\Users\lee\Anaconda3\envs\gopro\lib\site-packages\open_gopro\gopro.py", line 79, in _ensure_initialized_acquire_ready_semaphore
    ret = wrapped(*args, **kwargs)
  File "C:\Users\lee\Anaconda3\envs\gopro\lib\site-packages\open_gopro\gopro.py", line 681, in get
    request = requests.get(url)
  File "C:\Users\lee\Anaconda3\envs\gopro\lib\site-packages\requests\api.py", line 76, in get
    return request('get', url, params=params, **kwargs)
  File "C:\Users\lee\Anaconda3\envs\gopro\lib\site-packages\requests\api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "C:\Users\lee\Anaconda3\envs\gopro\lib\site-packages\requests\sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "C:\Users\lee\Anaconda3\envs\gopro\lib\site-packages\requests\sessions.py", line 655, in send
    r = adapter.send(request, **kwargs)
  File "C:\Users\lee\Anaconda3\envs\gopro\lib\site-packages\requests\adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPConnectionPool(host='10.5.5.9', port=8080): Max retries exceeded with url: /gopro/camera/stream/stop (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x0000025CB3461B20>: Failed to establish a new connection: [WinError 10060] A connection attem
pt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond'))
```

Would be glad to share my log file if more is helpful.  I think those are the predominant ones I saw on the console though

leeprevost, 2021-08-03:
@tcamise-gpsw - just a few more items to report in addition to the error log above.

1. one of the reasons I'm interested in power management is that the USB stream/control mode is not currently supported but believe it was previously and is also being supported in an undocumented way for the webcam mode.   I'd like to give a big vote to bring that back to the firmware as I could provide power and control the gopro via the USB port and not worry about battery life and battery monitoring.   Do you have input to the firmware engineers and where would be the appropriate place to register that input?
2. As agreed, battery testing should be a separate issue.   I'll be glad to start that. But, let me report that I'm finding the battery percent meter to be very difficult to interpret.  I've seen the AP go down at 25%, 20% and 9%.   And the full battery readings starting in the 70s and 80s (rather than showing "full".)   So, not sure how this meter interprets battery life (voltage level or a model) but i wonder if it needs to be reset or calibrated so as to provide more interpretive and predictive results.  
3. I wonder if AP is really being turned off or what is really going on with     `gopro.ble_command.enable_wifi_ap(False)`
 as when I disable it, I have difficulty in restarting it and then recovering it later to start a stream. 
 I'm not sure of the procedure to do this but it does not seem effective to just turn off and restart before streaming.   And as stated previously, the state doesn't seem to affect the battery discharge level/rate which is unintuitive.   Again, all of this is much less important to a product engineer who is considering designing in GoPro and OpenGoPro into their larger finished product if USB support is included in the interface.

tcamise-gpsw, 2021-08-04:
Regarding the errors, yeah I definitely need the complete log, i.e. the .log file.
Some of these look like bleak / Windows errors which I do not currently have a great idea of how to handle. Maybe disconnecting and retrying again or even resetting the computer's bluetooth. I'll look into it.

Regarding USB support...you can make an "enhancement" ticket on this repo. These will all get reviewed to see what should be added for a future release. I agree this sounds super useful and I can't think of any major roadblocks to adding it. Also, you're not the only person asking for this.

Battery testing...yeah let's split this off.

AP continuous enable / disable...this should certainly work. I haven't tested it but will try it out as part of this ticket.

leeprevost, 2021-08-05:
Hi Tim - -here is the link to the log file.  I think this log file captured several sessions and several attempts.    Let me know if this is what you needed.   
[link](https://www.dropbox.com/s/79qrjhtolhulpq6/gopro_test.log?dl=0)
thanks...

tcamise-gpsw, 2021-08-11:
Hey, so I'm going to close this out and merge what I currently have. Here is a summary of the remaining open items:
- _USB stream / control_: this has been confirmed to be coming in a future release. I don't have a date right now. If you want, you can make an enhancement request in this repo and I can update it when more information becomes available.
- _battery testing_: I've done some initial testing here and see the same result as you. After poking around, this isn't entirely unexpected. It appears that the command to turn off WiFi does not actually power off the WiFi chip. I'd like to get some more data here and, if it makes sense, take this internally for improvement. I'll create a discussion about this.
- _Bugs from your log_: I do see one worrisome WiFi timeout in that log file. It seems to be a bug on the camera side. However, I don't have a way to reliably reproduce this yet. I'm hoping it will show up during the battery testing from above.
- _Feature to go back to default WIFi in Python SDK after disconnecting from GoPro_: I'm going to hold off on this for now. Ultimately, I want to be able support connecting to multiple GoPro's which brings about it's own set of problems since we can only connect to one per WiFi adapter. So I need to some more thinking / redesigning about this.

closed 2021-08-11 by tcamise-gpsw

KonradIT, 2021-08-11:
>USB stream / control: this has been confirmed to be coming in a future release. I don't have a date right now. If you want, you can make an enhancement request in this repo and I can update it when more information becomes available.

Very cool to see GoPro adopting this feature!