gopro/OpenGoPro #21: Hero9 discoverable by bluetooth

Archimede55, 2021-07-22:
I'm using the Open Gopro program and I have a doubt about the possibility to discover the Hero9  by bluetooth.
On the website is written that it is discoverable in standby for 8h... I noticed that if it is connected to a power supply the camera is discoverable till the charge is finished then stop to be discoverable (also after 20 min if it start for example from 98%). Is there other conditions where the camera is no more discoverable? Can you confirm this behavior and why it does that?
 
Thanks

tcamise-gpsw, 2021-07-22:
Hello. I'm not entirely sure what you are asking here but I think it is a good idea to explicitly state the conditions when the camera is and is not discoverable. I'll look into this and reply here, then eventually update the BLE documentation.

Archimede55, 2021-07-23:
Thanks! Yes it was exactly the conditions when the camera is not discoverable. It seems that when the camera finished the charge (the red led go off) during the sleep is one of these conditions.

oliiiica, 2023-02-13:
Hi! I have also experienced this using the Hero11 Black. Is there an update for this?