gopro/OpenGoPro #24: Connect Hero9 to two different device

Archimede55, 2021-07-23:
The embedded system I'm developing use QT to manage the bluetooth on linux. It seems that when my system is connected to the Hero9 my iPhone is no more able to detect the camera with the quick app, viceversa when the hero9 is connected to the iPhone my system is not able to connect to the camera (not just a a matter of discoverable or not).

On the Polar application for the heart sensor there is a "flag" to allow the sensor to connect to multiple devices. Is there a similar command on the Hero9?

Thanks!

tcamise-gpsw, 2021-07-29:
Hello. What you are describing is the intended behavior. No, it is not currently possible (nor is there a configuration for) to maintain multiple simultaneous connections with the GoPro.  Out of curiosity, do you have a use case for this? It is theoretically possible since the GoPro supports BT 4.2 but would require some development on our side.

Archimede55, 2021-07-29:
Thanks for your answer! No at this moment I have no a specific case, my only concern was that if the customer has used the app and it is connected with it he was not able to connect the gopro to our system and he could think to a problem (simply because people don’t read the manuals when we will specify this thing).
But it could be useful if you use the mobile phone just for the screen to set the field of view but at the same time the gopro us connected to the system that for example manage the automatic start-stop precitding (this is my case)

closed 2021-07-29 by Archimede55