gopro/OpenGoPro #29: Investigate pairing on Linux

tcamise-gpsw, 2021-07-30:
Last I checked, there was a [bug](https://github.com/hbldh/bleak/discussions/503) in bleak when using BlueZ.

Need to check if this is still true and find a fix for it.

tcamise-gpsw, 2021-08-04:
There is currently an open PR in bleak for this: https://github.com/hbldh/bleak/pull/523

This is blocked until that is merged.

tcamise-gpsw, 2022-09-15:
I missed this somehow. This is pretty clever! I'll try it out and if it works will open a PR to get this into the Python SDK

tcamise-gpsw, 2022-10-19:
@pavlosharhan2 I finally got around to trying this today but I must be missing something because it is not working for me.

I tried debugging by calling `yes | bluetoothctl` from the command line. But this always just returns an error: "Invalid command in menu main: yes"

Any ideas?

pavlosharhan2, 2022-10-19:
> @pavlosharhan2 I finally got around to trying this today but I must be missing something because it is not working for me.
> 
> 
> 
> I tried debugging by calling `yes | bluetoothctl` from the command line. But this always just returns an error: "Invalid command in menu main: yes"
> 
> 
> 
> Any ideas?

@tcamise-gpsw are you running python script with sudo?

tcamise-gpsw, 2022-10-19:
Yes I am. 

Actually I ended up finding another way to do this with [pexpect](https://pexpect.readthedocs.io/en/stable/):

```
bluetoothctl = pexpect.spawn("bluetoothctl")
# bluetoothctl.logfile = sys.stdout.buffer
bluetoothctl.expect("Agent registered")
bluetoothctl.sendline(f"pair {handle.address}")
bluetoothctl.expect("Accept pairing")
bluetoothctl.sendline("yes")
bluetoothctl.expect("Pairing successful")
```

I like this and it seems to be working. I'll merge this soon.

tcamise-gpsw, 2022-10-26:
Closing this since it is now functionally working. The cleaner solution is still to implement this through bleak. If bleak ever supports this, we will adopt it.

closed 2022-10-26 by tcamise-gpsw