gopro/OpenGoPro #30: Media Mod & capturing video

Xav-v, 2021-08-02:
Hi all,

First, many thanks for such project.

To introduce the context:
 - I'd like to capture video in real time, processing & sending it over to webserver.
 - Few options studied & implemented (webcam, preview stream) but so far non answers my needs (needs stabilization, 1080p min, fully automated process as soon as camera plugged to a Rpi/connected to wifi)
 - Remaining option: I have the Media Mod, connected through HDMI to HDMI->USB capture, to a RPi4. Capturing is OK, but I still need some manual inputs to go to the "capturing video": the GoPro goes to "Review recent videos" page once Media Mod is connected through HDMI.

Is there any shortcut/status to use, to go directly to this video capturing mode, once the Media Mod is connected?

Otherwise, [this Issue](https://github.com/gopro/OpenGoPro/issues/10) seems a good option too.

KonradIT, 2021-08-03:
>Remaining option: I have the Media Mod, connected through HDMI to HDMI->USB capture, to a RPi4. Capturing is OK, but I still need some manual inputs to go to the "capturing video": the GoPro goes to "Review recent videos" page once Media Mod is connected through HDMI.

Try going to Settings > Mods > Media Mod. You should change it to "Live" instead of "Media"

Xav-v, 2021-08-06:
Thanks, but I don’t have such setting.
I only have: 
 - Max Lens Mod
 - Media Mod   - Standard Mic (read only, not selectable)

GoPro Hero 9
SW version: 1.60

KonradIT, 2021-08-06:
Ah, weird, try with the GoPro Labs Firmware. https://twitter.com/zaceubank/status/1409976950764965890?s=19

KonradIT, 2021-08-06:
(you'll need to scan the qr code after flashing the firmware)

Xav-v, 2021-08-07:
Works like a charm! Thanks a lot

closed 2021-08-07 by Xav-v