gopro/OpenGoPro #31: WiFi characteristics not found

sultanmyrza, 2021-08-08:
I was able to get untill https://gopro.github.io/OpenGoPro/tutorials/python/connect-wifi 

I use JavaScript instead of python (Ionic project)
I got it working on my ionic ios app. So far I was able to send commands to take picture and shutdown
```javascript
async sendCommand(scanResult: ScanResult, option: number) {
    const device = scanResult.device;
    const commandReqUUID = 'b5f90072-aa8d-11e3-9046-0002a5d5c51b';
    const service = '0000fea6-0000-1000-8000-00805f9b34fb';

    const command = option === 1
      ? [0x01, 0x05] // to shut down
      : [3, 1, 1, 1]; // to take picture
    const commandDataView = numbersToDataView(command);

    try {
      await BleClient.write(
        device.deviceId,
        service,
        commandReqUUID,
        commandDataView
      );
    } catch (error) {
      console.log(`error: ${JSON.stringify(error)}`);

    }
  }
```

Now I need to get WiFi SSID and PASSWORD of GoPro with bluetooh
```javascript
async getWifiSSID(scanResult: ScanResult) {
    const device = scanResult.device;
    const service = '0000fea6-0000-1000-8000-00805f9b34fb';

    try {
        const wifiApSsidUuid = `b5f90002-aa8d-11e3-9046-0002a5d5c51b`;
        const result = await BleClient.read(
          device.deviceId,
          service,
          wifiApSsidUuid,
        );
        console.log(`result: ${JSON.stringify(result.getUint8(0))}`);

      } catch (error) {
        console.log(`error: ${JSON.stringify(error)}`);

    }
  }
```

but I'm getting 

```
BluetoothLe read
Reject read|0000FEA6-0000-1000-8000-00805F9B34FB|B5F90002-AA8D-11E3-9046-0002A5D5C51B 
ERROR MESSAGE:  {"errorMessage":"Characteristic not found.","message":"Characteristic not found."}
```

tcamise-gpsw, 2021-08-10:
Hi. This is cool work that you're doing! If you get something working in Ionic, you should merge it into this repo as a sample project.

I have no idea how Ionic works but it looks like you might be using the wrong service UUID above. From the Open GoPro [BLE Spec](https://gopro.github.io/OpenGoPro/ble#services-and-characteristics), you should be using `b5f90001-aa8d-11e3-9046-0002a5d5c51b`

[image]

sultanmyrza, 2021-08-11:
@tcamise-gpsw Yep you were right I got it working now.
So far I was able to 

1. pair ionic app (ios) with GoPro (I will also add android later)
2. got WiFi ssid and password via bluetooth

TODO:
1. connect via wifi (for that I need to do native setup on ios first)
2. get media files from GoPro to ionic app (ios) later for (android)

@tcamise-gpsw sure if get things working I will share the experience and example code with some readme or short tutorial

tcamise-gpsw, 2021-08-11:
sounds good 👍

sultanmyrza, 2021-08-11:
Got it working finally 🤗. Was able to connect with Bluetooth than connect to go pro WiFi and list media files in go pro. 

After code cleanup I will share example ionic project 🤗