gopro/OpenGoPro #36: USB Stream/Control Enhancement Request

leeprevost, 2021-08-11:
Hey, so I'm going to close this out and merge what I currently have. Here is a summary of the remaining open items:
- _USB stream / control_: this has been confirmed to be coming in a future release. I don't have a date right now. If you want, you can make an enhancement request in this repo and I can update it when more information becomes available.
- _battery testing_: I've done some initial testing here and see the same result as you. After poking around, this isn't entirely unexpected. It appears that the command to turn off WiFi does not actually power off the WiFi chip. I'd like to get some more data here and, if it makes sense, take this internally for improvement. I've created a discussion about this: https://github.com/gopro/OpenGoPro/discussions/35
- _Bugs from your log_: I do see one worrisome WiFi timeout in that log file. It seems to be a bug on the camera side. However, I don't have a way to reliably reproduce this yet. I'm hoping it will show up during the battery testing from above.
- _Feature to go back to default WIFi in Python SDK after disconnecting from GoPro_: I'm going to hold off on this for now. Ultimately, I want to be able support connecting to multiple GoPro's which brings about it's own set of problems since we can only connect to one per WiFi adapter. So I need to some more thinking / redesigning about this.

_Originally posted by @tcamise-gpsw in https://github.com/gopro/OpenGoPro/issues/19#issuecomment-897030867_

KonradIT, 2021-09-16:
Well this is interesting 
[image]

tcamise-gpsw, 2021-11-22:
Quick update...this should be coming in January.

KonradIT, 2021-11-22:
That's great @tcamise-gpsw ! Will it be released for Kongs exclusively or Boilers too?

tcamise-gpsw, 2022-01-27:
Hello all. The latest release contains wired / webcam [documentation](https://gopro.github.io/OpenGoPro/http_2_0#webcam) as well as a [demo ](https://gopro.github.io/OpenGoPro/demos/csharp/webcam)so I'll close this.

closed 2022-01-27 by tcamise-gpsw