gopro/OpenGoPro #38: Installation and code execution

MokeyDGarp, 2021-08-18:
Hello, 

I am trying to control my two GoPro Hero9 via BLE by running a python script. Unfortunately I can't get any further with the documentation. I can neither connect the cameras nor execute commands. When I run the demo as in the documentation, I get the error message that 'GoPro()' does not exist. 

This is the code i used from the demo: 

[image]

I have installed the open-gopro with pip install open-gopro
What am i doing wrong?

Thank you very much for your help.

MokeyDGarp, 2021-08-18:
I'm sorry, the error message appears when i run this code:

[image]

Thank you very much for your help.

KonradIT, 2021-08-18:
Possible typo @tcamise-gpsw ? Should be called `GoPro` instead of `GoProBLE` in `.\demos\python\sdk_wireless_camera_control\docs\usage.rst`.

MokeyDGarp, 2021-08-18:
If I use GoPro() instead of GoProBle() I get the error message from my first post.

KonradIT, 2021-08-18:
Something weird with the system you're running if its not detecting any wifi interface.

tcamise-gpsw, 2021-08-18:
Yeah, it is a typo in the documentation: it should be GoPro(). I'll fix these for the next release.

@MokeyDGarp What OS are you using? Also, can you attach the log file? It should be a "xxx.log" in the directory where you are running your test.

MokeyDGarp, 2021-08-19:
Thank you. 

I am using Windows 10 Enterprise (64 Bit Version). 

It is the same error warning as in the other example. 

[image]

PLease find attached the .log file.

[photo.log](https://github.com/gopro/OpenGoPro/files/7012080/photo.log)

 Thank you very much for your help.

tcamise-gpsw, 2021-08-19:
Well this is certainly an interesting problem. What's happening is that your system language is apparently German:
```
   MainThread:                open_gopro.util:   89 07:56:51.519 DEBUG    | Send cmd --> netsh wlan show interfaces
   MainThread:                open_gopro.util:   97 07:56:51.765 DEBUG    | Receive response --> Der Dienst fr die automatische WLAN-Konfiguration (wlansvc) wird nicht ausgefhrt.
```

The WiFi Controller parsers are looking for specific (English) strings which is why they are not able to parse the response.

Honestly, I do not have any good ideas of how to handle this. I will add this into the documentation that the code only supports English. 

The quickest workaround for you is probably to set your system language to English. 

As an actual solution, we could start adding support for other languages but this would likely need to come from someone who speaks the language (i.e. not me).