gopro/OpenGoPro #39: Bleak async loops must be exited gracefully in order to prevent connection problems

tcamise-gpsw, 2021-08-19:
**Component**
Python SDK

**Describe the bug**
As per https://github.com/hbldh/bleak/issues/315#issuecomment-703641652, we are incorrectly exiting the demo applications by using Ctrl+C.

**To Reproduce**
Steps to reproduce the behavior:
1. Run any of the sample applications
2. Before they can complete, send Ctrl+C
3. See that the camera can no longer be found to connect to.  I have verified via BLE sniffer that this is because the original connection is still up.

**Expected behavior**
The connection should drop when the program exits.

**Hardware**
- Camera: Hero9
- Windows 10

tcamise-gpsw, 2021-08-19:
I think this can be handled in the GoPro class so that it is invisible to the user but need to do some testing to verify.

kimiyoshi, 2021-08-25:
I tested not only Win10, but also Mac OS.
Almost the same issue for both.

I want to instantly solve the issue for my urgent needs. 
Does someone have any fast remedy in mind?
Thanks in advance,

tcamise-gpsw, 2021-09-16:
This is fixed as of Open GoPro 0.6.0.

closed 2021-09-16 by tcamise-gpsw