gopro/OpenGoPro #63: RTMP control

DanielnetoDotCom, 2021-09-25:
Thanks for this app sample.

I tested the ionic demo and it works perfectly.

but I need to use the RTMP feature for the camera I read all the web and I could not find any documentation about it.

there is something I can read so can help me start a Livestream in a custom RTMP link?

thanks in advance

zsinba, 2021-10-21:
> Thanks for this app sample.
>
> I tested the ionic demo and it works perfectly.
>
> but I need to use the RTMP feature for the camera I read all the web and I could not find any documentation about it. [...]

Hello Mr.
what about your this issue?  did you have the solution to resolve this?
we have the same request, can you pls share your method about this?

Jens1985, 2022-01-17:
Any news here?

Seems the RTMP feature is a big secret.

hpwd, 2022-01-18:
Still needed here...

slifcoa, 2022-03-31:
Any update on this? I'd find this feature useful as well

DanielnetoDotCom, 2022-03-31:
This issue is just ignored by gopro. 

I suspect if they reveal how it works, it may be a security branch

KonradIT, 2022-03-31:
Not for the RTMP BLE call. For the RTMPS, it takes quite a long time to transmit the payload due to having to include a TLS security certificate in the BLE payload.

DanielnetoDotCom, 2022-03-31:
@KonradIT could you activate the  RTMP BLE using the apps?

KonradIT, 2022-03-31:
I showed how it works here: https://github.com/KonradIT/gopro-ble-py/issues/4#issuecomment-885487758

DanielnetoDotCom, 2022-03-31:
Thanks, I will check that.

tcamise-gpsw, 2022-09-15:
Hello. There are now steps in the BLE Spec to start [livestreaming](https://gopro.github.io/OpenGoPro/ble_2_0#live-streaming)

closed 2022-09-15 by tcamise-gpsw

DanielnetoDotCom, 2022-09-15:
Awesome, is there some code sample?

tcamise-gpsw, 2022-09-16:
Yep, there is a livestream command in the Python SDK [GUI](https://gopro.github.io/OpenGoPro/python_sdk/quickstart.html#api-gui-demo)

It ends up using [this code](https://github.com/gopro/OpenGoPro/blob/main/demos/python/sdk_wireless_camera_control/open_gopro/demos/gui/models.py#L347)

tcamise-gpsw, 2022-09-16:
Also, there is a simple script to use Docker to start a local RTMP server if you wanted to use this for testing [here](https://github.com/gopro/OpenGoPro/blob/main/demos/python/sdk_wireless_camera_control/tools/start_rtmp_server.sh)

exokinetic, 2023-05-11:
Just to add;

A slightly more refined version nginx-rtmp (local, lightweight, RTMP relay server) is available from the 'restreamer' project, search on github.

-Old version is deprecated but support ARMv6 mobile CPU architecture 
-New version has support for ARMv7 and ARMv8 for newer processors (and a lot of additional features for how lightweight it is)