gopro/OpenGoPro #68: Error launching photo/video demo

DG3D, 2021-09-28:
**Component**
What is the bug in? 
- python demo gopro-photo and gopro-video

**Describe the bug**
Launching those demo instantly throw this error : 
``` 
% gopro-photo                            
Traceback (most recent call last):
  File "/usr/local/bin/gopro-photo", line 8, in <module>
    sys.exit(main())
TypeError: main() missing 3 required positional arguments: 'identifier', 'log_location', and 'output_location'
```

**Hardware**
- Camera: Hero9
- MacOS BigSur

tcamise-gpsw, 2021-09-28:
Sorry about that. I've been testing by calling main directly so I missed that. I'll push out an updated open-gopro package tomorrow.

tcamise-gpsw, 2021-09-28:
Ok I've published version 0.6.2 which should have this fixed now

closed 2021-09-28 by tcamise-gpsw