gopro/OpenGoPro #72: Questions about streaming

sunwonlikeyou, 2021-10-06:
I'd like to make streaming service with wireless.
So  I want to know transfer speed of wifi and bluetooth.
And also received image type - jpg? or mp4? raw?
Does it only send images to url??

tcamise-gpsw, 2021-11-22:
Hello. This control is at a layer below the Open GoPro SDK and, at least for Bluetooth, is dependent on your computer (or whatever peer device you are using).

To summarize, you should (and only can) use WiFi for streaming. The throughput is at least good enough to work with the official GoPro Quik app.

tcamise-gpsw, 2021-11-22:
In terms of image format, you can choose either .jpg and / or raw (.gpr).

Here is a pretty good summary: https://havecamerawilltravel.com/gopro/photos-jpg-vs-raw-gpr/

tcamise-gpsw, 2021-11-22:
Closing since there are no actions to take.

closed 2021-11-22 by tcamise-gpsw