gopro/OpenGoPro #74: Delete media (feature?)

pyrat, 2021-10-09:
Hi,

Is it possible to delete media using the python connector?

I am testing taking many photos (every 5s) and would like to delete them to avoid filling up the SD card. I can see that its possible with the unofficial API but would be great if we can add support here. (maybe I have missed something if it is indeed possible.)

supperment, 2021-10-13:
I also need to know

Trimeego, 2022-03-29:
I am also interested in this feature.

KonradIT, 2022-03-31:
Seems to have been shipped in Hero10 v1.40 firmware.

`curl -x GET http://172.20.122.51/gopro/media/delete/file?path=100GOPRO/GOPR1464.JPG`

KonradIT, 2022-04-23:
Pushed a PR: https://github.com/gopro/OpenGoPro/pull/165

robertgalo, 2023-08-14:
> Seems to have been shipped in Hero10 v1.40 firmware.
> 
> Path: _/gopro/media/delete/file_
> 
> Param key: _path_
> 
> Param value: _XXXGOPRO/$file_
> 
> `curl -x GET http://172.20.122.51/gopro/media/delete/file?path=100GOPRO/GOPR1464.JPG`

Hello, tried with 11, but gives me nothing back but an Error 404.

tony-gutierrez, 2023-08-15:
Still nothing?

robertgalo, 2023-08-16:
> Still nothing?

Seems, that this all is not really maintained 😪. I mean, these are basic features (file handling).

tony-gutierrez, 2023-08-16:
Found out that there is a path for deleting a file on the 11 Black. 

Added this to my sdk http_commands.py:
```
    @http_get_json_command(endpoint="gopro/media/delete/file", arguments=["path"])
    def delete_media_file(self, *, file: str) -> GoProResp:
        """Delete a file on the camera.

        Returns:
            GoProResp: Empty object
        """
        return {"path": f"100GOPRO/{file}"}  # type: ignore
```

Also added settings for some missing things:
photo output mode control: INTERNAL_125
Beeps on/off: INTERNAL_87
Sharpness: INTERNAL_117

Would be nice if these were officially added to the SDK

The-Skunk, 2023-08-25:
> Found out that there is a path for deleting a file on the 11 Black.
>
> Added this to my sdk http_commands.py:
>
> ```
>     @http_get_json_command(endpoint="gopro/media/delete/file", [...]

Thank you, Tony. I implemented this in the http_commands.py (everywhere in the system) and it works.  In my opinion, this should be a standard feature. Hope the maintainer will read and then do this within the next 100 years ...

Coyne159, 2024-02-02:
Does anyone know if this work for a Go Pro Hero 10 black with firmware 1.5?

The-Skunk, 2024-02-02:
> Does anyone know if this work for a Go Pro Hero 10 black with firmware 1.5?

If you have your GoPro connected through USB, and you're able to access it through your browser (e.g. checking the state by: `http://172.2[ID-3].1[ID-2][ID-1].51:8080/gopro/camera/state `), you can try to execute this command in your browser if you have e.g. 100GOPRO as the current working directory set:

`http://172.2[ID-3].1[ID-2][ID-1].51:8080/gopro/version/gopro/media/delete/file?path=100GOPRO/[FILENAME].JPG`

Replace the values in the brackets by your settings (last digits of your Cam-ID by position -3, -2 and -1 (see manual and openGoPro documentation as well) and make sure that you've disconnected your camera from the file system where it hangs in normally by MTP.

tcamise-gpsw, 2024-04-16:
These endpoints have been added. There are 3 documented delete endpoints:
- [Delete All Files](https://gopro.github.io/OpenGoPro/http#tag/Media/operation/GPCAMERA_DELETE_ALL_FILES_ID)
- [Delete Single File](https://gopro.github.io/OpenGoPro/http#tag/Media/operation/OGP_DELETE_SINGLE_FILE)
- [Delete Grouped Media Item](https://gopro.github.io/OpenGoPro/http#tag/Media/operation/GPCAMERA_DELETE_FILE_GROUP)

closed 2024-04-16 by tcamise-gpsw

The-Skunk, 2024-04-19:
> These endpoints have been added. There are 3 documented delete endpoints:
>
>     * [Delete All Files](https://gopro.github.io/OpenGoPro/http#tag/Media/operation/GPCAMERA_DELETE_ALL_FILES_ID)
>
>     * [...]

Thank you for solving this all, we'll check all these and let you stay informed about the outcome ;-)

tony-gutierrez, 2024-04-19:
Be advised that the "delete all files" endpoint is very slow, even if there are no files to delete. 

Also, there is absolutly nothing in the state to indicate it is still processing. Encoding is not set to true???

The best I have found is to do a "delete all", pause for a while, then do a "get media list" and wait for that call to complete. It will not complete until the delete is done.