gopro/OpenGoPro #84: Bug Description

isht7, 2021-10-31:
**Component**
What is the bug in? 
In [this streaming demo](https://github.com/gopro/OpenGoPro/blob/main/demos/python/sdk_wireless_camera_control/open_gopro/demos/stream.py) in python sdk wireless camera control.

**Describe the bug**
I am using GoPro 7. When I run the stream.py, it keeps on searching for the camera, and is not able to find the camera. I have attached a screenshot below. I tried connecting the GoPro with my computer's bluetooth using the GoPro app on an android smartphone, but the laptop would not show up in the list of available devices in the app, even though it is visible on other devices.

**To Reproduce**
Steps to reproduce the behavior:
I run stream.py with a GoPro 7 camera.

**Expected behavior**
I expected the camera to be able to pair with my laptop and then be able to run stream.py successfully. 

**Screenshots**
[image]

**Hardware**
- Camera: Hero7
- Desktop: Dell XPS running Windows 10
- Smartphone: Samsung Galaxy S21

**Additional context**
Add any other context about the problem here.

tcamise-gpsw, 2021-11-22:
Hello. A couple notes here:
1. It appears that your camera is not advertising. As shown in log above, it is not being discovered. To advertise you need to tell the camera to start advertising as shown [here](https://gopro.github.io/OpenGoPro/tutorials/python/connect-ble#advertise):
2. The bigger problem is that the camera you are using (HERO 7) does not support Open GoPro. You need to use >= Hero 9 with the latest firmware.

tcamise-gpsw, 2021-11-22:
Closing since Open GoPro is not supported on Hero 7.

closed 2021-11-22 by tcamise-gpsw