gopro/OpenGoPro #87: Documentation:  BLE Characteristics permission not correct

XHMM, 2021-11-07:
**Component**
What is the bug in? 
- Documentation

**Describe the bug**
here [doc](https://gopro.github.io/OpenGoPro/ble_2_0#services-and-characteristics)  said `GP-0005` has `read/notify `permission, but I tested with my gopro10black, it's `read/indicate`

docs:
[image]

my results:
[image]

**To Reproduce**
connect to you gopro10black with some BLE debug tools to view

**Hardware**
- Camera: Hero10 black

tcamise-gpsw, 2021-11-22:
Thanks for pointing this out. I will update this.

tcamise-gpsw, 2022-01-27:
Hi. This is fixed in the latest [docs](https://gopro.github.io/OpenGoPro/ble_2_0#services-and-characteristics):

closed 2022-01-27 by tcamise-gpsw