gopro/OpenGoPro #91: Set date/time (missing?)

developer-appdam, 2021-11-18:
Hi,

I keep trying to read the documentation where it states that it's possible to set the date/time but I really don't see how to set the date/time. Is this actually possible? And if so, how can we set the date/time of the GoPro through the API? Everything that I can see so far are just GET calls.

Thanks

KonradIT, 2021-11-18:
True, needs to be added to this wiki.

It's specified in the unofficial docs: https://github.com/KonradIT/goprowifihack/blob/master/Bluetooth/bluetooth-api.md#command-requests

tcamise-gpsw, 2021-11-22:
Hello. Yeah, we should not have mentioned it without the API documented. In terms of official support, it is coming soon. That is:
1. A new open gopro endpoint will be added to set time via WiFi
2. Documentation will be added to show how to set time via BLE

developer-appdam, 2021-11-24:
Perfect! For now, I will try to implement the current "unofficial" method

tcamise-gpsw, 2022-01-27:
Hi. This has been added in the latest firmware release / documentation. Note that the datetime status has been removed. So now there are datetime get and set commands for both BLE and HTTP.

closed 2022-01-27 by tcamise-gpsw