gopro/OpenGoPro #92: Connect to existing wifi

aL3891, 2021-11-20:
**Is your feature request related to a problem or use case? Please describe.**
I'd like to control and see live previews from several gopros from a computer, however since only wifi AP is supported, this is not possible

**Describe the solution you'd like**
Enable gopros to connect to an exsisting wifi network

**Describe alternatives you've considered**
I suppose its possible to use several wifi adapters but that is very cumbersome

**Additional context**
Having the ability to connect many gopros to a single network would unlock a lot of scenarios and make others alot easier

tcamise-gpsw, 2021-11-22:
Hello. This unfortunately is a design constraint of the current WiFi network topology. 

You can, of course, control the cameras serially by connecting / disconnecting WiFi as needed to send commands. But that will incur some serious latency due to the numerous connection overheads.

So I do certainly see the benefit to your request; I'm just not sure if it feasible. I will take this up internally and get back to you.

KonradIT, 2022-02-12:
Connecting to an existing WiFi AP already supported by GoPro cameras for something called **Camera as a Hub** (codename CAH), which is the daemon tasked with sending media to the GoPro cloud servers. Having the opengopro API server / gpControl API server run alongside it and listen to requests on the GoPro exposed IP (0.0.0.0) would accomplish this. Not too far off from the `gopro/camera/control/wired_usb` API call.

Wakaij1n, 2023-09-13:
Hi guys !

I'm also super interested in that feature. Any plan about it ?

Cheers!

tcamise-gpsw, 2023-12-06:
This is new supported via the [COHN ](https://gopro.github.io/OpenGoPro/http_2_0#camera-on-the-home-network-cohn)(Camera on the Home Network) feature: