gopro/OpenGoPro #102: ImportError: cannot import name 'Final' from 'typing' (/usr/lib/python3.7/typing.py)

MadMahdi, 2021-12-08:
Hello Guys,

Thanks for this tuto it's really great, I was able to connect through BLE and was able to start the shutter and to stop it.

When it comes to the Wifi I'm stuck at one step. 

Until this step it's fine :
```
OpenGoPro/demos/python/sdk_wireless_camera_control $ pip show open-gopro
Name: open-gopro
Version: 0.7.0
Summary: Open GoPro API and Examples
Home-page: https://gopro.github.io/OpenGoPro/python_sdk/
Author: Tim Camise
Author-email: tcamise@gopro.com
License: MIT
Location: /home/pi/.local/lib/python3.7/site-packages
Requires: betterproto, bleak, construct, protobuf, requests, rich, typing-extensions, wrapt
Required-by: 

```

But when I'm arriving to this one:
```

~/OpenGoPro/demos/python/sdk_wireless_camera_control $ gopro-photo
Traceback (most recent call last):
  File "/home/pi/.local/bin/gopro-photo", line 5, in <module>
    from open_gopro.demos.photo import entrypoint
  File "/home/pi/.local/lib/python3.7/site-packages/open_gopro/__init__.py", line 6, in <module>
    from open_gopro.gopro import GoPro
  File "/home/pi/.local/lib/python3.7/site-packages/open_gopro/gopro.py", line 14, in <module>
    from typing import Any, Dict, Final, Optional, Type, Callable, Union, Generic, Pattern
ImportError: cannot import name 'Final' from 'typing' (/usr/lib/python3.7/typing.py)

```
It's not working because of the ImportError.

On my Raspberry Pi4 I had Python 3.7 installed but i did the upgrad to Python3.8 abd it's default python. SO it's supposed to work

What should I do ? Do you have any ideas ?

Thank you

Mahdi 

- Camera: Hero9
- Rapsberry Pi4
- Python3.8

KonradIT, 2021-12-08:
You need to uninstall open gopro completely and install it using python3.8

https://docs.python.org/3/library/typing.html#typing.Final

MadMahdi, 2021-12-08:
Hello Konrad thanks a lot for you help and you previous help with your
contribution on the OpenGoPro .... stupid quesiton. I tried to uninstall
OpenGoPro with the commande :

sudo apt-get --purge remove OpenGoPro ... it doesn't seem to work.

Thank you

tcamise-gpsw, 2021-12-16:
Hello. Can you verify that you are actually using 3.8 via `python --version` (or however you are calling python). It should show 3.8. It looks like you are somehow still calling 3.7.

If that is the case, you'll need to call the 3.8 python using it's absolute path or update your PATH variable to prefer 3.8 over 3.7.

tcamise-gpsw, 2022-01-05:
Closing due to inactivity.

closed 2022-01-05 by tcamise-gpsw