gopro/OpenGoPro #104: Not able to connect to the internet while connected to the Gopro

pavloshargan, 2021-12-21:
I am not able to use 2 wi-fi dongles to connect to the GoPro and to the internet at the same time.
After I connect to the internet, I create a GoPro object, which connects to the GoPro via one of 2 wi-fi devices, that has the biggest score in the network system. It is impossible to change this score manually, so we can't predict which wi-fi device will be used. So It randomly picks one of the wi-fi dongles therefore often connects via a dongle that was previously connected to the router and is already in use. It would be great to select the wifi device, or make the GoPro object use one of the wi-fi devices, which is not connected to any network.

Solution1:
The solution I see is to add the wifi_device_name parameter to the GoPro initialization parameters, so we can select the wi-fi device to use.
Solution2(optional):
Another alternative is to scan which wi-fi devices are not connected to any networks, and use one of them.

I have already proposed a pull request with Solution1 implemented.

tcamise-gpsw, 2022-01-03:
Thanks for looking into this, it is certainly a valid use case. I like Solution 1 also. I made a PR to get this working (as well as update the tests. Can you check it out to see if it works for you?

tcamise-gpsw, 2022-01-03:
I've been meaning to do a major cleanup of the Wifi adapter so I have done that in the pull request linked above. It required some bumping of test module versions which in turn required updating some non-related files. Anyway, it is all done in #108 and is at least working on Windows. I'll do some more testing on Mac and Ubuntu before merging.

tcamise-gpsw, 2022-01-06:
> [EXTERNAL]
> Hi,
> Thanks for your work, would be happy to see this feature in the new SDK
> release.
> Would like to clarify if I missed something: could I help somehow from my
> side?

Hello. You don’t necessarily need to do anything but it would be good if you can test my changes<https://github.com/gopro/OpenGoPro/pull/108> to make sure it solves your problems.

I’ve done a good amount of testing myself so if you don’t have time to test, I’ll end up merging the PR regardless.

-Tim

pavloshargan, 2022-01-07:
Will make a tests tomorrow and reply back to you

closed 2022-01-15 by tcamise-gpsw