gopro/OpenGoPro #106: GPS position data

ShaynAli, 2021-12-23:
**Is your feature request related to a problem or use case? Please describe.**
The GoPro has an impressively powerful GPS, if we, as developers, could leverage the GPS this would open up an entirely new world of possibilities. To name a few:
- Use of the GoPro in science projects, including wildlife tracking or weather balloon retrieval, where the GoPro provides stunning footage AND the journey can be re-traced.
- Use as a precision timer in high performance driving, e.g. at road racing events, drag races, or rally events. Note that the precision provided by phones is too low for this, but GoPros offers an 10Hz+ signal which is sufficient.
- Use as an adventure tracking tool, to retrace steps of an adventure.
There are many, many more applications of the tech that other more creative developers than me will envision.

**Describe the solution you'd like**
I would like the GoPro's GPS latitude and longitude to be provided in the API at a 10Hz+ rate, if possible.

**Describe alternatives you've considered**
A connected phone can provide some data, but the quality is typically so low that it greatly hampers many possible applications. Additionally, in many applications it is not possible or practical for a phone to remain connected.

**Additional context**
None at this time.

tranzmatt, 2023-04-05:
Was this feature ever enabled?  I would find it very useful.

ThiagoMattosLabTrans, 2025-07-24:
I too would like if this data became available.