gopro/OpenGoPro #117: BLE (Bluetooth) connection not working on MacOS Monterey (V12.1)

PipGrylls, 2022-01-24:
On MacOS (V12.1) running the following code.

`from open_gopro import GoPro`
`gopro = GoPro()`
`gopro.open()`
`print(gopro.is_ble_connected)`

Will fail on line 3 `gopro = GoPro()`. 

`INFO     14:29:06.754 Scanning for GoPro [A-Z0-9]{4}        bleak_wrapper.py:120`
`         bluetooth devices...`                                                   
`Failed to find a device in 5 seconds. Retrying #1`
`INFO     14:29:11.818 Scanning for GoPro [A-Z0-9]{4}        bleak_wrapper.py:120`
`         bluetooth devices...`                           

Tried with passing target or leaving blank. This seems to be related to the issue in Bleak generally.
See the open issue with Bleak [here](https://github.com/hbldh/bleak/issues/720).

tcamise-gpsw, 2022-01-27:
Hello. I have a fix for this on the main branch here but have not made a release to PyPi yet.
I have a little bit more cleanup / testing to do but will make a release at some point in the next week.

tcamise-gpsw, 2022-02-03:
I have merged the fix for this to main and it has been published to PyPI.
FYI it looks like Apple has [fixed ](https://github.com/hbldh/bleak/issues/720#issuecomment-1025017418) the original bug.
I'm going to leave the scan for service UUID's since we have no need to find any devices that don't have the GoPro service UUID.

closed 2022-02-03 by tcamise-gpsw

PipGrylls, 2022-02-04:
Good stuff thanks for this