gopro/OpenGoPro #118: Change stram quality

pavloshargan, 2022-01-27:
The API doc says, that the quality of the stream could be different, up to 1080p. Now I can get only 720p 30Fps via this command:
http://10.5.5.9:8080/gopro/camera/stream/start

I am able to change the quality of the stream in the gopro-quick mobile app, so I suppose there is some url to change the resolution of the stream, that wasn't mentioned in the API. Like this:

http://10.5.5.9:8080/gopro/camera/stream/start?res=1080p

The question: how can I change the resolution of the stream?

KonradIT, 2022-01-27:
Do you mean the quality of the RTMP livestream? That can only be changed when initiating a livestream and has to be done over BLE.

pavloshargan, 2022-01-27:
> Do you mean the quality of the RTMP live stream? That can only be changed when initiating a Livestream and has to be done over BLE.

Does GoproHero10 support different types of streams?

I mean UDP stream, which I start via http://10.5.5.9:8080/gopro/camera/stream/start, and read on udp://0.0.0.0:8554

pavloshargan, 2022-01-27:
@KonradIT 

Ok, I suppose I have understood, that it could be 2 types of the stream:

Preview:
-Streaming to GoPro local network
-720p by default
-starts/stops via wifi command http://10.5.5.9:8080/gopro/camera/stream/start(stop) and could be accessed on udp://0.0.0.0:8554

Livestream:
-Streams directly from the GoPro to the internet (to rmpt server for instance)
- could be 1080p/720p/480p
-could be configured and started via Bluetooth commands (don’t know which ones)

I am mostly interested in streaming the preview to Gopro's local network.
Now I am able just to start and stop it. But I am really wanting to set its resolution to 1080p. Is that possible?, If yes - then how can I do It?

tcamise-gpsw, 2022-11-15:
As mentioned above, the Preview Stream end point now takes resolution and fov arguments.

closed 2022-11-15 by tcamise-gpsw

alexivaner, 2025-01-03:
Hi everyone, I think I could not find where we can change preview stream. The res and fov argument only available for webcam mode. Also I want to confirm, can we change the FPS for preview stream? Currently I always get 1920x1080p with 30FPS. Is it possible to make 60 FPS? I did not find the possibility to change this on preview stream.