gopro/OpenGoPro #119: Bug Description

pavloshargan, 2022-01-27:
Bug in  Demos (python sdk, etc)

 Nmcli command fails connecting to the GoPro AP, if the Gopro ssid contains spaces

[image]

fixed manually adding the quotes: (OpenGoPro/demos/python/sdk_wireless_camera_control/open_gopro/wifi/adapters/wireless.py line 408 class: Nmcli0990Wireless, method: connect)

```
nmcli dev wifi connect \"{ssid}\" password {password} ifname {self._interface}
```

[image]

Suppose this issue could happen using other drivers classes as well

tcamise-gpsw, 2022-01-27:
Hello. Thanks for pointing this out. I'm in the middle of updating that Wifi driver so will add this and test it out.

tcamise-gpsw, 2022-02-03:
This has been fixed on main but not yet published to PyPi

closed 2022-02-03 by tcamise-gpsw