gopro/OpenGoPro #121: Handle unexpected BLE disconnect

pavloshargan, 2022-01-30:
I'm always frustrated when getting the "Ble connection terminated unexpectedly" error 

It happens quite often and doesn't allow develop the robust applications.

I'm not into details of Bleak, so can't bring an idea of how to manage prevent/manage such disconnects.

Does anyone have an idea how to handle that?

mlaity, 2022-04-19:
I am also continually have this issue:  BLE connection terminated unexpectedly.  This is on a Raspberry Pi 4, fully updated, Python v3.9.2, connecting to a Hero10 with v1.40 firmware.  I am unable to catch this exception within my own code for some reason.  It is happening within gopro.py, within the _disconnect_handler method of the GoPro class.  Sure would be nice if that disconnect handler could handle an automatic BLE reconnect.
I can typically connect to the GoPro, parse statuses, sometimes send a Start/Stop command or two, maybe even get a video downloaded successfully, but inevitably terminates unexpectedly.

ERROR:root:A message handler raised an exception: Connection terminated: BLE connection terminated unexpectedly..
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/dbus_next/message_bus.py", line 668, in _process_message
    result = handler(msg)
  File "/usr/local/lib/python3.9/dist-packages/bleak/backends/bluezdbus/client.py", line 1040, in _parse_msg
    self._disconnected_callback(self)
  File "/usr/local/lib/python3.9/dist-packages/open_gopro/gopro.py", line 600, in _disconnect_handler
    raise ConnectionTerminated("BLE connection terminated unexpectedly.")
open_gopro.exceptions.ConnectionTerminated: Connection terminated: BLE connection terminated unexpectedly.

tcamise-gpsw, 2022-05-17:
Hello all. While there is nothing that can be done about the unexpected connection termination, I agree that you should certainly be able to catch the exception and handle it gracefully as desired.

I thought that I had actually made that change at some point but perhaps not. I'll look into this and make the change if needed.

tcamise-gpsw, 2022-05-18:
So what's happening here is that, when the connection is unexpectedly dropped, the exception is being raised from the thread that is running the Bleak controller. Therefore, it will never reach the main application thread.

There is currently way of passing internal messages like this between threads. I could make the GoPro's output queue more generic and pass it there but I don't necessarily like this.

My proposed solution is to add an optional unexpected_disconnect_callback argument to the GoPro class that will be called when this occurs. The main thread can then do with this what it chooses.

pavloshargan, 2022-05-18:
I’ve ended up with using the Bluetooth only for the connection process, and using wifi http api for all other purposes

mlaity, 2022-05-18:
I've been going down the path of having a very simple BLEDisconnect callback for the BleakClient to use.  That callback just sets a global variable that then the BLE Keep-Alive process checks at each loop, and will attempt to reconnect if it is no longer connected via BLE.  Putting a BLE Reconnect directly in the callback was failing for me for various reasons.  I also have pushed a lot of my commands back over to WiFi, with BLE only for the initial connection, pairing, enabling WiFi.

I very much look forward to testing out your proposed solutions!  Thanks for taking up this issue!

closed 2022-05-26 by tcamise-gpsw