gopro/OpenGoPro #123: Can't connect to the Gopro wifi AP

pavloshargan, 2022-02-01:
- Demos (python sdk)

Sometimes the password for the command was just cut (for example if there is < symbol). I figured It out checking the password in the ubuntu wi-fi settings. I've fixed it adding the quotes to the cmd command. But Still I get this error (Nmcli0990Wireless connect):
```
Error: Connection activation failed: (7) Secrets were required, but not provided
```

 OpenGoPro/demos/python/sdk_wireless_camera_control/open_gopro/wifi/adapters/wireless.py line 408

[image]

closed 2022-02-01 by pavloshargan

fypsoftware39, 2023-10-01:
what was the fix?

pavloshargan, 2023-10-03:
@fypsoftware39  the problem was that I had to put the SSID and PASSWORD to double quotes when executing cmd commands. Sometimes password or said contains special characters.