gopro/OpenGoPro #135: "400 Bad request" when calling /media/info with USB connection

SebastianSchoeps, 2022-02-13:
**Component**
What is the bug in?
- Demos (python sdk, etc)
- Tutorials
- Documentation
- Specs (BLE , HTTP)

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect GoPro with USB
2. Call `GET /media/list` --> works
```
{
    "id": "2087...",
    "media": [
        {
            "d": "100GOPRO",
            "fs": [
                {
                    "n": "GX010012.MP4",
                    "cre": "1644663618",
                    "mod": "1644663618",
                    "glrv": "276391386",
                    "ls": "-1",
                    "s": "4004385314"
                },
            ]
        }
    ]
}
```
Call `GET /media/info?path=100GOPRO/GX010012.MP4` --> `400 Bad Request`

**Expected behavior**
Media info according to https://gopro.github.io/OpenGoPro/http_2_0#commands-quick-reference

**Hardware**
Hero 10 Black

**Additional context**
Hey, I started playing around with the API and a new GoPro and was very happy how quick this started to work with USB. However, I cannot get the media info to work. Is this supposed to work with USB?

Thanks,
Sebastian

tcamise-gpsw, 2022-08-24:
Hi it seems like what you are doing is correct. I just tested this out and it works, i.e.:

```
$ curl http://172.24.180.51:8080/gopro/media/info?path=100GOPRO/GX010283.MP4
{"cre":"1641444050","s":"26836553","mahs":"1","us":"0","mos":[],"eis":"0","pta":"1","ao":"stereo","tr":"0","mp":"0","ct":"0","rot":"0","fov":"4","lc":"0","prjn":"9","gumi":"89b20f55564e2eb4998bf361ef58e265","ls":"2764605","cl":"0","avc_profile":"1","profile":"153","hc":"0","hi":[],"dur":"5","w":"1920","h":"1080","fps":"60000","fps_denom":"1001","prog":"1","subsample":"0"}
```

Just to avoid any obvious errors, can you please post your complete commands and responses for your media list and get media info commands?

SebastianSchoeps, 2022-09-02:
Thanks for your answer. I still have the issue.

```
$ curl http://172.22.108.51:8080/gopro/media/list                           
{"id":"3974853737622611528","media":[{"d":"100GOPRO","fs":[{"n":"GX010142.MP4","cre":"1662138786","mod":"1662138786","glrv":"1262941","ls":"-1","s":"17108086"},{"n":"GX010143.MP4","cre":"1662138789","mod":"1662138789","glrv":"2101522","ls":"-1","s":"29263262"}]}]}

$ curl http://172.22.180.51:8080/gopro/media/info?path=100GOPRO/GX010142.MP4
zsh: no matches found: http://172.22.180.51:8080/gopro/media/info?path=100GOPRO/GX010142.MP4
```

SebastianSchoeps, 2022-09-02:
My bad with the quotes, sorry about that.

```
$ curl 'http://172.22.108.51:8080/gopro/media/list'                               
{"id":"3974853737622611528","media":[{"d":"100GOPRO","fs":[{"n":"GX010142.MP4","cre":"1662138786","mod":"1662138786","glrv":"1262941","ls":"-1","s":"17108086"},{"n":"GX010143.MP4","cre":"1662138789","mod":"1662138789","glrv":"2101522","ls":"-1","s":"29263262"}]}]}

$ curl 'http://172.22.108.51:8080/gopro/media/info?path=100GOPRO%2FGX010142.mp4'
{}

$ curl -I 'http://172.22.108.51:8080/gopro/media/info?path=100GOPRO%2FGX010142.mp4'
HTTP/1.1 400 Bad Request
Content-Type: application/json
Content-Length: 3
```

tcamise-gpsw, 2022-09-15:
I think your last error is due to case sensitivity in the file type. The file in the media list is uppercase (MP4) while you are sending lowercase (mp4)

Can you retry?

SebastianSchoeps, 2022-09-19:
Thanks for your help. It works now. I don't know what went wrong (I tried all kinds of combinations of uppercase / lowercase before).

closed 2022-09-19 by tcamise-gpsw