gopro/OpenGoPro #136: GoPro Hero 9 open GoPro API connection issue

koloyyee, 2022-02-14:
```
open_gopro/gopro.py", line 660, in _write_characteristic_receive_notification
logger.trace( # type: ignore
AttributeError: 'Logger' object has no attribute 'trace'
```

tcamise-gpsw, 2022-02-15:
Hello. Which part of the tutorial are you at?

Also, it looks like you somehow ended up with a newer version of the Open-GoPro Python SDK then what is specified in the requirements.txt (0.7.1) in the tutorials. Did you follow [these steps](https://gopro.github.io/OpenGoPro/tutorials/python/connect-ble#tutorials-package) to install the package? It is imperative to use that specific version because the tutorials are not updated at the same cadence that the [Open GoPro Python SDK](https://pypi.org/project/open-gopro/) is updated.

koloyyee, 2022-02-17:
Thanks for replying, I was following the Open GoPro Python SDK.
Because when I was following the [official tutorial](https://gopro.github.io/OpenGoPro/tutorials/python/connect-ble#tutorials-package) it only `INFO:root:Found 0 matching devices.` even though I connect to the GoPro 9 with my MacBook Air M1.

And I do notice for the BLE 1.0 mentioned about GoPro 9, is it one of the reasons? thank you again

wemat, 2022-04-09:
hello i have the same error ('Logger' object has no attribute 'trace') and I installed the package (pip) and run the sample code bellow. I tried it on my macbook pro and on a raspberry pi with the same result.

[image]

tcamise-gpsw, 2022-06-06:
Regarding the logger.trace error: this has been fixed in open-gopro versions >= 0.9.1 which the tutorials are now using.

Regarding the Invalid Open GoPro Version error, this is expected as the open-gopro Python SDK only supports version 2.0. There is no reason to use 1.0 at this point as 2.0 is available on all cameras that support 1.0.

Therefore closing this as there are no open issues to handle.

closed 2022-06-06 by tcamise-gpsw