gopro/OpenGoPro #137: GoPro 8 Support

zawminoo-i2r, 2022-02-14:
I can connect wifi with GoPro 8. But I cannot stream to GoPro 8. Is OpenGoPro working for GoPro 8?

KonradIT, 2022-02-14:
This is a complicated answer, the OpenGoPro spec for Bluetooth is the same as the commands available online prior to the release of OpenGoPro that were obtained via bluetooth packet capture. So you can use OpenGoPro BLE code to control a HERO8. In order to do WiFi operations (control, view media, get a live preview) you need to use the gpControl/Smarty API, which is documented unofficially: https://github.com/konradit/goprowifihack

tcamise-gpsw, 2022-03-04:
Closing since Hero 8 is not technically supported as mentioned above.

closed 2022-03-04 by tcamise-gpsw

tcamise-gpsw, 2023-01-15:
I have no idea what you are talking about but please read the [code of conduct](https://github.com/gopro/OpenGoPro/blob/7adb3262f73fb9f5e592087df6ad0b1a6067ccb4/CODE_OF_CONDUCT.md)  for this repository as your comment is in violation of it.

Consider this a warning and further such behavior will result in a ban.