gopro/OpenGoPro #143: Can't auto-assing interface. None found <- I dont know what mean is..

sunwonlikeyou, 2022-03-04:
**Component**
What is the bug in?
- Demos (python sdk, etc)

My laptop conntected GoPro with wifi.  Cause I checked 10.5.5.9/DCIM/Videos/
So I ran the code.. 
but..
```
gopro = GoPro()
```

Error 
Exception: Can't auto-assing interface. None found
[image]

and also available interfaces were empty.

what is the wifi interface and what should i do..??

**Hardware**
- Camera: Hero10, etc.
- Windows
- IPhone 12 Pro

tcamise-gpsw, 2022-03-04:
Hello. The wifi interface is the wifi adapter that your PC uses to connect to the camera. You can set this manually but by default it attempts to be automatically discovered. I'll need to see the detailed log to see what is wrong with the autodiscovery here so do the following:
1. enable debug logging in the util module. That is, set `"open_gopro.util": logging.DEBUG` in the `default_modules` map in `util.py`
2. run the Wifi Connect [demo](https://gopro.github.io/OpenGoPro/python_sdk/quickstart.html#wifi-demo): `gopro-wifi`
3. Post the contents of the `gopro_wifi.log` file here

sunwonlikeyou, 2022-03-07:
Thank you for your reply.

And This is the log file 

[image]
.

tcamise-gpsw, 2022-05-26:
Hi. Can you post the entire log. We need to see what is happening in the beginning where the interface can't be auto-assigned.

sunwonlikeyou, 2022-06-20:
This is the entire log.[](url
[image]
)

tcamise-gpsw, 2022-08-01:
I'm sorry but that is still not the entire log. Just attach the entire .log file. Specifically I need to see the response to the interface query command. For example, on Windows mine looks like this:

```
   MainThread:11:55:56.035 util.py                                   311 DEBUG    | Send cmd --> netsh wlan show interfaces
   MainThread:11:55:56.102 util.py                                   319 DEBUG    | Receive response --> 

There is 1 interface on the system: 

    Name                   : Wi-Fi

    Description            : TP-Link Wireless USB Adapter

    GUID                   : 093d8022-33cb-4400-8362-275eaf24cb86

    Physical address       : 98:48:27:88:cb:18

    State                  : disconnected

    Radio status           : Hardware On

                             Software On

    Hosted network status  : Not available

```

tcamise-gpsw, 2022-09-15:
Closing due to inactivity. Will improve logging and documentation of this situation in #218

closed 2022-09-15 by tcamise-gpsw

isdito, 2023-01-23:
Hello,

I Have this interface

```
Hay 1 interfaz en el sistema:

    Nombre                   : Wi-Fi
    Descripción            : TP-Link Wireless USB Adapter
    GUID                   : 2a588865-9192-44d3-9dfa-56c24084f2ec
    Dirección       : 98:48:27:b7:7a:74
    Tipo de interfaz         : Principal
    Estado                  : desconectado
    Estado de radio           : Hardware Activado
                             Software Activado

    Estado de la red hospedada: No disponible
```
I bought a gopro 11, to try to connect via bluetooth or wiffy. I want to be able to record a certain time and send something very simple to the computer that is taking me a long time to focus on what I have to do. On my other gopro5 it is very simple.

I don't know what the wiffy interface is and how I have to put the command.

gopro-wifi --wifi_interface Wi-Fi

this not works