gopro/OpenGoPro #148: Support USB control in Python API

aallaire91, 2022-03-18:
Is there a plan to support USB control in the python api? I'm experiencing intermittent connection issues with the wifi interface and would like a more reliable connection via USB.  It seems like much of the functionality provided by the gopro.wifi_command interface could be used, but there needs to be additional logic to use the correct IP address for USB and I'm guessing the code handling wifi-connection would need to be side-stepped somehow.  I could probably hack something together myself, but wanted to check if an official plan to support this feature was already in progress.

dyldata, 2022-04-05:
Wanting to also bring this up, did you have any luck @aallaire91 - or is there any plan for this to be a thing? I also have the same question and would like to control upwards to 5 cameras at once via this method.

lucasmediaflow, 2022-04-10:
Hi, same here. Any plans for this?
Trying to get something working myself, but so far no idea where to even begin

KonradIT, 2022-04-10:
Discussed over here: https://github.com/gopro/OpenGoPro/discussions/146

tcamise-gpsw, 2022-05-17:
Hello. I don't currently plan on adding USB to the Python SDK. Once the SDK is more mature / changing less I may look into it.

tcamise-gpsw, 2022-12-17:
FYI this has been merged and will be released as part of the 0.12.0 release (later today)

closed 2022-12-17 by tcamise-gpsw