gopro/OpenGoPro #153: GoPro MAX - WiFi connection fails if SSID is unchanged (default)

CallumBasham, 2022-03-25:
**Component**
- Tutorials: Tutorial 5 Python SDK:  https://gopro.github.io/OpenGoPro/tutorials/python/connect-wifi
- GoPro QUIK, makes me believe its a camera fault not necessarily connection! 

**Describe the bug**
OpenGoPro Python Tutorial, after Bluetooth successfully pairs, the device attempts to connect to the GoPro's Access Point, in this step it fails. 

HOWEVER, if you do the exact same within the GoPro QUIK app, after pairing at the Bluetooth level, it asks if you wish to change the camera SSID/Name, if you leave as default:
_The GoPro fails to connect to Wi-Fi, same as with OpenGoPro Tutoral._
If you change the SSID, 
_The Go Pro and device connect perfectly!_ 

**To Reproduce**
Steps to reproduce the behavior:
1.  Enable connections on the GoPro
2.  Set to 2.4 ghz instead of 5
3. Click connect QUIK or Remote (depending on your choice)
4. Bluetooth should connect as expected
5. [on QUIK] leave the SSID/Name the same when it asks
6. Enable & connect to the GoPro WIFI
7. It should fail to connect with no error messages presented. 
8. At Step (5), change the SSID, then the connection should work as expected! 

**Expected behavior**
Expect that the end device should be able to connect to the Go Pro's Wi-Fi access point.

**Hardware**
- Camera: GoPro MAX
- Ubuntu/ BlueZ (same components as suggested within the tutorial) 
- GoPro QUIK on OnePlus7 Pro Android 

**Additional context**
As a temporary fix, if there is a method within the Python OpenGoPro tutorial to change the GoPro's SSID via Bluetooth, that should allow me to continue with my work, as it appears to work like so on the GoPro QUIK app. 
(I did not see this particular function documented and would greatly appreciate any assistance :) ).

tcamise-gpsw, 2022-08-01:
Hello, the GoPro Max is not one of the [supported cameras ](https://gopro.github.io/OpenGoPro/ble_2_0#supported-cameras)for Open GoPro so this behavior is effectively undefined. Can you reproduce this on Hero 10 / 9?

CallumBasham, 2022-08-01:
I am no longer seeking a solution to this - I have opted to use a different manufacturer. Thanks for the response however!

closed 2022-08-01 by CallumBasham