gopro/OpenGoPro #155: convert 360 video

soheer, 2022-03-28:
Hello,

We have GoPro Max camera that records a video with 360 extension (.360) .
Using GoPro Quik App (mobile) we can save into our phone with mp4 extension but it remains a 360 view.
How can we convert the recorded video into standard video (mp4) without loosing 360 feature capability) using code or command line .

Thanks.

demiantres, 2022-04-09:
You can use the GoPro Player, which is available for macOS and Windows, for this.

soheer, 2022-04-11:
we know that, but our aim is to run those features through code (python) so the **code** ( no external software )will connect to the camera, record video and download it as shown in this api with extra feature which is to convert it into mp4 while  preserving the 360 capabilities.

tcamise-gpsw, 2022-05-17:
Hello. This is currently out of the scope of Open GoPro which is simply an interface (USB / WiFi / Bluetooth) to control the camera.

In any case I did find this project which may perhaps be of use to you: https://github.com/trek-view/MAX2sphere

closed 2022-05-17 by tcamise-gpsw