gopro/OpenGoPro #157: BLE API - Malformed multi-packet responses

cjmconie, 2022-04-04:
**Component**

- BLE API

**Describe the bug**

When making a single request for multiple status values, where the response payload is larger than 20 bytes (i.e. the response is sent as multiple packets which must be accumulated), the response packets are malformed and cannot be parsed.

This does not happen when the request is for fewer status values where the response is sent in a single packet.

**To Reproduce**

Send a request to 0076 with payload (here in decimal) [11, 19, 70, 68, 35, 86, 69, 97, 96, 11, 8, 19]. As an example, here are the responses from 7 separate requests:

```text
// send request 1 (parsing fails)
out: 81-00-00-00-01 // cont. header (packet 3) [Bad]
out: 81-00-00-00-01 // cont. header (packet 3) [Bad]
out: 81-00-00-00-01 // cont. header (packet 3) [Good]

// send request 2 (parsing fails)
out: 20-29-13-00-08-01-00-0b-01-00-13-01-00-23-04-00-00-0c-a7-44 // ext-32 header (packet 1) [Good]
out: 81-00-00-00-01 // cont. header (packet 3) [Bad]
out: 81-00-00-00-01 // cont. header (packet 3) [Good]

// send request 3 (parsing succeeds)
out: 20-29-13-00-08-01-00-0b-01-00-13-01-00-23-04-00-00-0c-a7-44 // ext-32 header (packet 1) [Good]
out: 80-01-00-45-01-01-46-01-3a-56-01-00-60-04-00-00-03-e8-61-04 // cont. header (packet 2) [Good]
out: 81-00-00-00-01 // cont. header (packet 3) [Good]

// send request 4 (parsing fails)
out: 80-01-00-45-01-01-46-01-39-56-01-00-60-04-00-00-03-e8-61-04 // cont. header (packet 2) [Bad]
out: 81-00-00-00-01 // cont. header (packet 3) [Bad]
out: 81-00-00-00-01 // cont. header (packet 3) [Good]

// send request 5 (parsing fails)
out: 20-29-13-00-08-01-00-0b-01-00-13-01-00-23-04-00-00-0c-a7-44 // ext-32 header (packet 1) [Good]
out: 81-00-00-00-01 // cont. header (packet 3) [Bad]
out: 81-00-00-00-01 // cont. header (packet 3) [Good]

// send request 6 (parsing fails)
out: 80-01-00-45-01-01-46-01-39-56-01-00-60-04-00-00-03-e8-61-04 // cont. header (packet 2) [Bad]
out: 81-00-00-00-01 // cont. header (packet 3) [Bad]
out: 81-00-00-00-01 // cont. header (packet 3) [Good]

// send request 7 (parsing fails)
out: 20-29-13-00-08-01-00-0b-01-00-13-01-00-23-04-00-00-0c-a7-44 // ext-32 header (packet 1) [Good]
out: 81-00-00-00-01 // cont. header (packet 3) [Bad]
out: 81-00-00-00-01 // cont. header (packet 3) [Bad]
out: 81-00-00-00-01 // cont. header (packet 3) [Bad]
out: 81-00-00-00-01 // cont. header (packet 3) [Bad]
out: 81-00-00-00-01 // cont. header (packet 3) [Good]
```

The only response which is parseable came from request 3. All other requests appear to have malformed payloads.

**Expected behavior**

All response payloads should have 3 packets similar to the repose of request 3:

```text
out: 20-29-13-00-08-01-00-0b-01-00-13-01-00-23-04-00-00-0c-a7-44 // ext-32 header (packet 1) [Good]
out: 80-01-00-45-01-01-46-01-3a-56-01-00-60-04-00-00-03-e8-61-04 // cont. header (packet 2) [Good]
out: 81-00-00-00-01 // cont. header (packet 3) [Good]
```

**Hardware**
- Camera: Hero 9 Black (v1.6)
- Custom iOS App leveraging CoreBluetooth. I managed to reproduce it by extending this repo's Swift demo code too.

tcamise-gpsw, 2022-05-17:
Hello. I haven't actually parsed your bytes here but what you are describing is extremely common functionality so I'm guessing it is more likely that you are misunderstanding the parsing scheme.

Before I go through to check the data here, can you read through this relevant [tutorial](https://gopro.github.io/OpenGoPro/tutorials/python/parse-ble-responses#parsing-multiple-packet-tlv-responses) to verify that your understanding is correct?

Thanks

tcamise-gpsw, 2022-08-01:
I just tested this with 50 iterations and do not see any problems. Every single time I see:
```
| Sending get settings : 0b:13:46:44:23:56:45:61:60:0b:08:13 to Query
| Received response on Query Response: 20:29:13:00:08:01:00:0b:01:00:13:01:00:23:04:00:00:11:d9:44
| Received response on Query Response: 80:01:00:45:01:00:46:01:2f:56:01:00:60:04:00:00:03:e8:61:04
| Received response on Query Response: 81:00:00:00:00
```

Also I maintain that it is extremely likely for this to be broken. So I will close this as "can't reproduce."

For you own testing, you might want to try sending the command with Light Blue on your mobile device to remove your app from the picture.

closed 2022-08-01 by tcamise-gpsw