gopro/OpenGoPro #162: Can't run ble_connect.py !

smallworld-network-wupeng, 2022-04-20:
**Component**
What is the bug in?
- Demos (python sdk, etc)

**Describe the bug**
System : MacOS 12.3.1 (21E258)
Python : 3.8.13
Error:
```
(gopro) wupeng@MacBook-Pro-2 tutorial_1_connect_ble % python3 ble_connect.py
INFO:root:Scanning for bluetooth devices...
zsh: abort      python3 ble_connect.py
```
Please help!

tcamise-gpsw, 2022-06-06:
Hello. I just updated the tutorials to fix some bugs. Can you please try this again?

Please post the entire log if your are seeing a problem.

Also, make sure you are using a supported [camera and firmware version.](https://gopro.github.io/OpenGoPro/ble_2_0#supported-cameras)

tcamise-gpsw, 2022-08-01:
closing due to inactivity

closed 2022-08-01 by tcamise-gpsw