gopro/OpenGoPro #164: Allow for zoom setting through API.

EddyCelis, 2022-04-23:
**Is your feature request related to a problem or use case? Please describe.**
It is now possible to get the zoom status (percent of zoom) through the API.
There is no documented way to SET the current zoom on the camera.

**Describe the solution you'd like**
Add a setting to allow for changing the zoom setting.

**Describe alternatives you've considered**
None.

**Additional context**
It seems the feature does not exist in the firmware, as the Quik app does not allow to change the zoom either.

KonradIT, 2022-04-23:
You can. 

`/gopro/camera/digital_zoom?percent=50` (from: https://gopro.github.io/OpenGoPro/http_2_0)

EddyCelis, 2022-04-24:
Should have specified: through Bluetooth.

shubham9436, 2022-06-13:
Is it possible to set Zoom level through Bluetooth?

KonradIT, 2022-06-13:
Most likely not, why would GoPro allow for zooming in over BLE when the app uses WiFi for preview, and thus, zooming is done over WiFi as well.

EddyCelis, 2022-06-14:
No it is not.

I flagged it as an enhancement request.

tcamise-gpsw, 2022-06-14:
Hello, I'm also wondering why you want this. Can you provide a use case or some reasoning on what value this brings? Why is not acceptable for this command to only be exposed via HTTP?

EddyCelis, 2022-06-14:
In our application we allow for control of several cameras and every
camera is reset to a pre-configured preset at app startup. We wanted to
add the capability to also set the zoom independent of the preset zoom
setting, so that when the app sets up the camera at startup, the zoom
setting is also adjusted. If this is only possible through http, the wifi
connection will have to be connected (and acknowledged by the user) for
every camera that is initialized.

tcamise-gpsw, 2022-06-14:
Since the actual effect of setting the zoom isn't visible to the user until, at minimum, WiFi is connected, could you just set the zoom to your desired default value immediately after the WiFI is first connected? And more generally, do the same for any settings that only are relevant when WiFi is connected? 

I suppose you incur a non-zero startup delay to set these defaults but it is only for the first WiFi connection and is almost certainly not noticeable.  

From the other perspective, it is certainly possible to expose this setting via BLE. AT this time, these types of HTTP-specific settings are not exposed via BLE so this potentially opens the floodgates to exposing all settings, etc via BLE. Perhaps this is actually a good thing. 
I will take it up for discussion internally but I wouldn't expect anything in the immediate future.

EddyCelis, 2022-06-14:
This is for a specific application where the zoom setting will change depending on other settings (non-GoPro) in the application at startup and while running. We could have 2 or more cameras running at the same time that could all have a need to change the zoom. As this is a ‘moving’ kind of application (car, airplane, boat), we absolutely want to make sure there is no distraction (e.g. need to acknowledge the wifi) while running.

But it is not an absolute deal breaker for the app, just a nice-to-have, so we have commented out the code for that feature.

 

Is there a list with enhancement requests? We have some other nice-to-haves 😉

lwp-joel, 2025-05-12:
I would just like to add a small +1 this issue.

Our use case is to perform sanity checks on each of our GoPros prior to each take to ensure each are shooting at the correct settings. In case digital zoom does not match our desired value (0), it would be great to be able to automatically set it to the correct value over BLE.