gopro/OpenGoPro #169: command/setup/date_time API gives 500 response code

harshmdr-devslane, 2022-05-05:
**Describe the bug**
{"version": "6.00","path_info": "command/setup/date_time","query_string": "p=%16%05%05%0665","error_code":-3,"error_msg": "500 Internal Server Error","function": "gpcontrol_cgi_handler","line": 190}

**Hardware**
- Camera: Hero8 Black.

tcamise-gpsw, 2022-05-17:
Hello. I'm not sure where you're getting that API from. Here is the relevant documentation from the Open GoPro 2.0 [quick reference](https://gopro.github.io/OpenGoPro/http_2_0#commands-quick-reference):

[image]

tcamise-gpsw, 2022-06-06:
Closing since the supported command is working.

closed 2022-06-06 by tcamise-gpsw