gopro/OpenGoPro #172: Getting Response Timeout Running the Tutorial

aobeidhp44, 2022-05-18:
**Component**
What is the bug in?
- Tutorials

**Describe the bug**
I am following the instruction on OpenGoPro Tutorials, the only one worked for me is ble_connect.py. 
Basically, when running the script, I am getting a connection and notification enabled done message but then as if it get stuck awaiting for a response from the camera. Moreover, I tried running some commands from the gopro cli and same isse,

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the tutorials modules folder
2. Run python [name of script you want to run]

**Expected behavior**
Getting a response back and disconnect from client

**Screenshots**
[image]

[image]

**Hardware**
- Camera: Hero10, etc.
- Intel Based MacBook Pro

**Additional context**
Add any other context about the problem here.

Yuhan-Lu, 2022-06-06:
I met the same problem with the Gopro Max... I still couldn't figure out this problem. 
At the same time, I also tried [https://github.com/KonradIT/gopro-ble-py](this unofficial one). 
I found out that both BLE solutions stuck at the write_gatt_char method that it would never return correctly, especially in those lines in tutorial 2 ble_command_enable_analytics.py

`    # Write to command request BleUUID to enable analytics
    logger.info("Enabling analytics...")`
   ` event.clear()`
    `await client.write_gatt_char(COMMAND_REQ_UUID, bytearray([0x01, 0x50]))`
    `await event.wait()  # Wait to receive the notification, response never get a result in my case`

tcamise-gpsw, 2022-06-06:
Hello. I just updated the tutorials (5 minutes ago) and tested them all. Can you please try again?

If you are still seeing an error, please post the log.

Also ensure that you are using a supported [camera / firmware version](https://gopro.github.io/OpenGoPro/ble_2_0#supported-cameras).

Yuhan-Lu, 2022-06-06:
Really appreciate for your work and reply, unfortunately, I still met the same problem as before that it get stuck awaiting response from the camera.

tcamise-gpsw, 2022-06-06:
Please post the entire log. Also what camera / firmware are you using?

Yuhan-Lu, 2022-06-06:
Thanks for your reply! I am currently working on the Gopro Max, I understand it's not officially supported, and I would be really appreciate if you could give a hint. According to my sketchy understanding, most of the BLE-based command operations are still very similar for Max and Hero 10. They both write to GP0072 and wait for notification of 0073 (for some control commands), and basically based on Bleak. 
For example for sleep, I think we would write
`"b5f90072-aa8d-11e3-9046-0002a5d5c51b"` to camera.

When I tried to have a better log for you just now, I changed this line to get some printouts.
`test = await client.write_gatt_char(COMMAND_REQ_UUID, bytearray([0x01, 0x05]), True)`
It just worked great if I add the return argument for write_gatt_char function!
I would be really appreciate if you could share some thoughts!!!
Failure when without changing write_gatt_char:
[image]

Success:
[image]