gopro/OpenGoPro #173: 4k streaming over usb

cleoag, 2022-05-18:
camera can encode into 4k, why not allow streaming in 4k mode via usb?

KonradIT, 2022-05-18:
Streaming over USB was originally meant for use with the GoPro webcam PC software, which just needed >720p stream. Don't think the camera can handle such bandwidth over it's USB -- and you would need a very capable USB3 compatible machine to read at such speeds anyways.

cleoag, 2022-05-19:
> 

logitech brio 4k can stream 4k via usb, so there is no problem with bandwidth if you can control encoder before sending resulting stream via usb. we are streaming 4k60fps with 120Mbs over usb in custom developed cameras, i dont see any problems Gopro10 can do this as well. 

lots of videobloggers will be happy with 4k

lozo2010, 2023-09-05:
Is there any chance they are going to implement this?