gopro/OpenGoPro #177: GoPro Cloud Service API

tenninebt, 2022-05-24:
**Is your feature request related to a problem or use case? Please describe.**
GoPro Quik for desktop is not maintained anymore. I would like to create an alternative with more features. To be able to do so, I need to have an API to be able to use the GoPro Cloud Service (For users with a Plus Subscription).

**Describe the solution you'd like**
An API giving access to the media stored in the cloud (List/Get Metadata/Download/Upload/Update/Delete)

**Describe alternatives you've considered**
None available that I know of.

**Additional context**
Please refer to first question/answer.

KonradIT, 2022-05-25:
Already reverse engineered: https://github.com/dustin/gopro-plus

tenninebt, 2022-05-26:
I saw it, but it is not an official support, it might become incompatible or obsolete anytime. Moreover, GoPro might block the access anytime as it is not meant to be a public API.

tcamise-gpsw, 2022-11-15:
The current scope of Open GoPro is limited to Camera interface API. So this is currently out of scope but I will leave this open in case this changes.

sid-geospoc, 2023-01-02:
Hi devs,

Is there anything on your roadmap to add not only GoPro Cloud, but also any cloud such as AWS/Azure/Google , basically any URL as required by the user?

This is in relation to using any cloud url to upload data from GoPro to any personal cloud.

tcamise-gpsw, 2023-01-04:
Hi. No, the scope has not changed since my last comment: Open GoPro is still solely focused on communicating with the camera.

Fougl, 2023-07-15:
> Already reverse engineered: https://github.com/dustin/gopro-plus

Does this still work today?