gopro/OpenGoPro #179: Feature Description

user2003j, 2022-05-25:
RTMP streaming is making a lot of latency when u are trying to stream to a webinterface on your own home Server it has a latency about 13 seconds because the web interface has to encode the stream to hls protocol to display it in a Webbrowser. Since flash player is not used anymore you cannot display a stream in your Webbrowser via rtmp protocol.

The solution would be that the GoPro is able to support multiple streaming formats such as hls since rtmp protocol is not supported. If the GoPro could support a few more streaming protocols that are newer such as hls it would be possible to stream with a much lower latency and display the stream in a Webbrowser with out a delay of 13 seconds it makes the streaming feature unusable for streaming in a lot of cases

i tried to conquer this problem by encoding the stream with multiple options but the encoding to hls causes a delay of 13 seconds or more.

i would be quite happy if the GoPro gets an update so more streaming protocols would be supported. If you have other solutions to fix this problem I am open to these as well. Thank you.

developer-appdam, 2022-06-13:
How do you actually stream from the GoPro? Because I haven't had any luck yet, it seems that what I get from the camara can't be decoded..

tcamise-gpsw, 2023-12-06:
Hello. Still no HLS support now or in the planned future. There is however now RTSP support via the (optionally wireless) [webcam streaming](https://gopro.github.io/OpenGoPro/http_2_0#webcam).

Regarding how to stream, I would recommend looking at the [demos](https://gopro.github.io/OpenGoPro/demos) which several streaming examples