gopro/OpenGoPro #184: Camera Stuck In Buggy State - Commands Broken 500 Internal Server Error

codercoder117, 2022-06-06:
**Component**
BLE and HTTP software on Hero10 camera.

**Describe the bug**
A bunch of functionality on the camera from bluetooth no longer workers and the commands return invalid responses or a 500 internal server error. The camera seems to be stuck in a weird state such that some commands work fine, while other commands return `02:05:02` on BLE or `500 Internal Server Error` on HTTP. I am currently unable to access camera physically, so being able to debug or fix remotely would be hugely invaluable.

At the moment, I can connect to the camera via a bluetooth device. I am able to disable or enable the WiFi AP successfully and it returns the correct response '`02:17:00` via bluetooth API. I am also able to communicate with the HTTP API on the camera with some commands such as `/gopro/media/list` which works properly and also initiating downloads of media files works as well.

However, on the bluetooth API, there are many commands that do not work and send the incorrect response code such as the sleep command which return the response code `02:05:02` instead of powering off the camera. Also if I send the shutter ON or shutter OFF commands, those also send incorrect response codes and do not function properly.

In addition to that, the HTTP API also does not work with certain commands such as the `/gopro/media/delete` command which returns a 500 Internal Server Error, and also the `/gopro/camera/state` command which returns a 500 Internal Server Error.

Summary: Some commands are broken on both the BLE and HTTP API's of the camera, while some commands work on both of them.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to camera via bluetooth device
2. Set the camera setting so that it will only record a maximum of 60 minutes
3. Sent a Shutter-ON command
4. Wait 75 minutes and then send the Shutter-OFF command (it should return an incorrect response code since the camera already stopped recording)
5. Camera is now stuck in the bugged state described above

**Hardware**
- Camera: Hero10, software v1.4.20

bfreeland-gpsw, 2022-11-03:
Hi @codercoder117,

How often are you encountering this issue? Are you able to reproduce it consistently? Does it persist after power cycling the camera?

ImJulester, 2022-12-19:
Has there been any progress fixing this bug? I get this error every time i restart my pc and try to send a HTTP command. It says USB connected until i try to send a HTTP command and then i get the 500 internal server error. The only fix is to reconnect the usb c cable but for the system we are building we can't access the camera.

I use the Gopro 11 mini.

robertgalo, 2023-08-08:
Update to the previous issue (above):

Following we found out to "resolve" the issue by a workaround (we used the webbrowser for simplification) and didn't do any python stuff until now:

1) Fact/Impact of the tests a) - c) from above: The USB connection "p=1" gets "somehow" destroyed ... - some commands work, some not - generally the cam through its USB connection is in a indifferent state.

2) we could reproduce the error as well we can reproduce then the solution (workaround), which is as the following (sequence of http-commands through browser):

start (cam through USB connected, notation: URL --> http response, 200 = OK, 500 = Error):

http://172.2X.1YZ.51:8080/gopro/camera/control/wired_usb?p=1 --> 200
http://172.2X.1YZ.51:8080/gopro/camera/shutter/stop --> 200
http://172.2X.1YZ.51:8080/gopro/camera/state --> 200 (plus JSON output)
http://172.2X.1YZ.51:8080/gopro/camera/shutter/start --> 200
http://172.2X.1YZ.51:8080/gopro/camera/shutter/stop --> 200

Check, video recorded ok!

Now we unmounted the cam through the file explorer
http://172.2X.1YZ.51:8080/gopro/camera/control/wired_usb?p=1 --> 200
http://172.2X.1YZ.51:8080/gopro/camera/shutter/stop --> 200
http://172.2X.1YZ.51:8080/gopro/camera/state --> 200 (plus JSON output)
http://172.2X.1YZ.51:8080/gopro/camera/shutter/start --> **500**
http://172.2X.1YZ.51:8080/gopro/camera/shutter/stop --> 200

We can repeat now this from above, the camera will stay/reply when starting always 500 ...  

Solution: Before we 
http://172.2X.1YZ.51:8080/gopro/camera/control/wired_usb?p=1 --> 200
we do a
http://172.2X.1YZ.51:8080/gopro/camera/control/wired_usb?p=0 --> 200
and then again
http://172.2X.1YZ.51:8080/gopro/camera/control/wired_usb?p=1 --> 200
http://172.2X.1YZ.51:8080/gopro/camera/shutter/stop --> 200
http://172.2X.1YZ.51:8080/gopro/camera/state --> 200 (plus JSON output)
http://172.2X.1YZ.51:8080/gopro/camera/shutter/start --> 200
http://172.2X.1YZ.51:8080/gopro/camera/shutter/stop --> 200

everything works fine again

pavlosharhan2, 2023-08-11:
having the same issue. Response 500 when requesting a state every 5 seconds. Getting an error once in 1-2 minutes. Haven't seen this problem much, but it looks like the more i use the camera, the more frequently i experience the bug. Year ago, half a year ago I've seen this bug only a couple of times, but now i see it all the time.

Faulty 1.50 firmware?

robertgalo, 2023-08-14:
> having the same issue. Response 500 when requesting a state every 5 seconds. Getting an error once in 1-2 minutes. Haven't seen this problem much, but it looks like the more i use the camera, the [...]

Yes, probably. It seems not to be an issue of OpenGoPro.

tony-gutierrez, 2024-04-19:
I have found that on Ubuntu 18, the usb connection is dropped and reconnected, necessitating you to resend the usb control command before shutter start. You can check the usb control state by looking at status 116 in the state response. 

If you try to do shutter commands without usb control, you get a 500