gopro/OpenGoPro #188: Feature Description

peetCreative, 2022-06-14:
**Is your feature request related to a problem or use case? Please describe.**
I want to write a linux service running in background using the OpenGoPro API using Python.
But my programm is asking me for sudo rights on the command line. Disabling Wifi with `GoPro(enable_wifi=False)`  won't work, because WifiModule is instantiated in GoPro anyways.
  

**Describe the solution you'd like**
1. I configure my linux using /etc/network/interfaces or similar to automatically connect the right way to the GoPro AP if available.
2. When disabling Wifi, I'm not asked for any password anymore.

**Describe alternatives you've considered**
Passing the password into the file starting the service.. That means in some file in `/etc/...` I have to state the plaintext password.

tcamise-gpsw, 2022-06-14:
This seems very reasonable as there is no reason to request sudo if you do not want to use WiFi. 

I'll look into adding this, it should hopefully be pretty trivial.

closed 2022-06-14 by tcamise-gpsw

tcamise-gpsw, 2022-06-14:
Hi I just merged a change that will:
- add a `sudo_password` argument to the `GoPro` class
- not instantiate the WiFi interface if `enable_wifi` is false
- add a command line argument to the demos to accept the sudo password from stdin

So you have lots of options now. If you don't need wifi, you can set `enable_wifi` to false and you will never be prompted for the password. Or you can send the password via stdin, for example:
`cat mypasswordfile.txt | gopro-photo --password`