gopro/OpenGoPro #194: Can't auto assign interface

JulesVtr0, 2022-06-20:
Hi, i'm currently trying to devellop a multi camera network with a modulable number of gopro. Except that when i launch my code an error occured. 

PS : the BLE part work great, I'm just stuck when connecting to the WiFi AP

Here's the error log:
```
Traceback (most recent call last):
  File "c:\Users\jules.vautier\camera\connect_WIFI_AP.py", line 131, in <module>
    asyncio.run(main(args.timeout))
  File "C:\Users\jules.vautier\AppData\Local\Programs\Python\Python310\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Users\jules.vautier\AppData\Local\Programs\Python\Python310\lib\asyncio\base_events.py", line 646, in run_until_complete
    return future.result()
  File "c:\Users\jules.vautier\camera\connect_WIFI_AP.py", line 106, in main
    WiFi_client =       await connect_wifi(ssid, password)
  File "c:\Users\jules.vautier\camera\connect_WIFI_AP.py", line 93, in connect_wifi
    wifi = Wireless()
  File "C:\Users\jules.vautier\AppData\Local\Programs\Python\Python310\lib\site-packages\open_gopro\wifi\adapters\wireless.py", line 67, in __init__
    self.interface = interface  # type: ignore
  File "C:\Users\jules.vautier\AppData\Local\Programs\Python\Python310\lib\site-packages\open_gopro\wifi\adapters\wireless.py", line 187, in interface
    self._driver.interface = interface  # type: ignore
  File "C:\Users\jules.vautier\AppData\Local\Programs\Python\Python310\lib\site-packages\open_gopro\wifi\controller.py", line 112, in interface
    raise Exception("Can't auto-assign interface. None found.")
Exception: Can't auto-assign interface. None found.
```
Thanks in advance !

tcamise-gpsw, 2022-08-08:
Hello. This means that the Wifi module can not automatically find a suitable WiFi interface. The quickest solution here is to pass in the actual interface that you want to use. To see the list of interfaces, you can do the following depending on the OS:
- Windows: `netsh wlan show interfaces`
- (Most) *nix: `nmcli dev`
- MacOS: `networksetup -listallhardwareports`

then pass it in using the top level interface parameter of the [GoPro](https://gopro.github.io/OpenGoPro/python_sdk/api.html#gopro) class.

If you want to find out why it is failing to find a suitable interface, you can:
1. Run the default gopro-wifi demo
2. look at the resulting gopro_demo.log after it fails. You're specifically looking for the response to the interface discovery command from above. If you post the **entire** log here I can help take a look.

As a side note, you're not the first person to get confused here so I need to update the logging and docs to make this more clear. That work will be done [here](https://github.com/gopro/OpenGoPro/issues/218).

tcamise-gpsw, 2022-09-15:
Hi.  Are you able to find your list interfaces or provide the requested log?

tcamise-gpsw, 2022-11-16:
Closing this as #216 has added more detailed error messages as well as documentation on how to handle the wifi interface.

closed 2022-11-16 by tcamise-gpsw

Fougl, 2023-07-17:
Hi, I am not able to assign interface. even manually. On my RPI if i run nmcli dev I get Error:NetworkManager is not working. I know my interface name is wlan0 as always. If iI try to sudo systemctl start NetworkManager, then the whole interface goes down due to dhcpcd not running error. If I enable back the dhcpdc then i return to the previous error