gopro/OpenGoPro #197: Can't connect to GoPro HERO 9/10 over HTTP/UDP, Systray App works/connects no problem

GriffinRU, 2022-06-24:
**Component**
What is the bug in?
- Documentation
- Specs (BLE , HTTP)

**Describe the bug**
Following online documentation, cannot establish connection with camera over HTTP protocol nor as simple as VLC -> udp://@0.0.0.0:8554. GoPro app works great (from system tray). Would be nice to have more in-depth description how packets created and consumed, and why address specified by 'zeros' not exact.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware**
- Camera: Hero9, etc.
- Desktop if applicable [e.g. Windows]
- Smartphone if applicable [e.g. iPhone6]

**Additional context**
Add any other context about the problem here.

KonradIT, 2022-06-24:
You need to query the GET URL to start preview: `/gopro/camera/stream/start`

Make sure your firewall allows UDP packets over 8554.

Also, this is not the place to create issues regarding general product and software issues/feature requests.

GriffinRU, 2022-06-24:
Ok, where I should report bugs/features... this link was given to me directly from GoPro
So, firewall, ok my development IDE is on firewall exempt list and all ports are available. GoPro app also works properly, so port 8554 approved -> what's the catch?
How I can start stream if I cannot open socket?
Probably missing something obvious... hint in right direction would be great.
-Artur

GriffinRU, 2022-07-06:
Ok, new software (v.1.2.1.813) update and some tweaks on windows firewall (not related to port, rather to network type 'Unidentified network') and 'YES' - success.
By the way, still plenty small bugs related to video start/stop, and moving forward what should I use "0.0.0.0" or actual device IP?

closed 2022-07-06 by GriffinRU