gopro/OpenGoPro #202: Camera stream not visible

developer-appdam, 2022-06-27:
**Component**
What is the bug in?
Demos (python sdk, etc)
Tutorials

**Describe the bug**
Can't get the liveview to work. We are currently implementing it in Swift and we can't get anything close to a stream or images.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to the Gopro(using the way explained in the tutorials)
2. Set the GoPro to stream with the command /gp/gpControl/execute?p1=gpStream&c1=restart
3. Play the stream using MJPEGStreamLib(for Swift) using the following URL udp://127.0.0.1:8555/gopro?pkt_size=64(also tried with 10.5.5.9)

**Expected behavior**
A visible stream when playing the UDP stream

**Hardware**
- Camera: Hero9 and Hero 10, etc.

KonradIT, 2022-06-27:
1. You're not using the OpenGoPro Start Stream URL, rather the unofficial one
2. The port is 8554, and the IP is 10.5.5.9, no substream needed.

KonradIT, 2022-06-27:
Correct UDP URI.

bfreeland-gpsw, 2022-10-26:
The correct URL for the UDP stream is udp://@0.0.0.0:8554 and the format should be an MPEG transport stream. If you're having issues try verifying the stream is playing correctly by opening it as a network stream in VLC.

tcamise-gpsw, 2022-11-15:
@developer-appdam Is this still a problem? Were you able to verify the network stream in VLC as suggested above?

developer-appdam, 2022-11-15:
I will check it out this week. Had no time before, so let me come back to you on this.

tcamise-gpsw, 2023-12-06:
Closing due to inactivity and inability to reproduce

closed 2023-12-06 by tcamise-gpsw