gopro/OpenGoPro #204: unable to send any commands

z0mb0funk, 2022-07-08:
**Component**
What is the bug in?
- Tutorials

**Describe the bug**
After successfully completing Tutorial 1, I am unable to send any commands to any of my GoPros.

**To Reproduce**
Steps to reproduce the behavior:
Run any script other than ble_connect.py

**Expected behavior**
Varies. Right now I'm just trying to progress through the tutorials. Expected behavior is defined in Tutorial 2. I connect to the GoPro send the command, but never progress past "setting shutter to on" or "putting camera to sleep" (etc, I've tried all scripts)

**Screenshots**
[image]
[image]

**Hardware**
- Camera: Hero9 - 01.72.70& Hero10 - 01.42.70. 
- macOS (intel chip)

**Additional context**
Add any other context about the problem here.

z0mb0funk, 2022-07-08:
There are no timeouts or errors. It will just indefinitely sit at "sending command", even if I turn off the camera.

tcamise-gpsw, 2022-08-01:
It is because you have put the camera to sleep.

z0mb0funk, 2022-08-01:
The second screenshot is an example of that command being tested. If you look at the first you'll see another example. I tried every example in the wiki and got the same result. It isn't because I'm putting the camera to sleep. The camera should send a response to ack the command and that isn't happening here.

tcamise-gpsw, 2022-08-08:
Sorry I was reading the log incorrectly.

I am able to replicate this (only on Mac). The solution is to add a `response=True` parameter to all of the `write_gatt_char` method calls. This is a nasty little bug that happened when the bleak version got upgraded which updated the default value of this argument only on Mac.

I have a branch where I am working on fixing this. I'm also going to remove the open-gopro Python module dependency since upgrading this has the side effect of upgrading the bleak version.

tcamise-gpsw, 2022-08-08:
This should be working now but let me know if there are any problems

closed 2022-08-08 by tcamise-gpsw

z0mb0funk, 2022-08-09:
Seems to be working as intended now. I just downloaded the new tutorial files and it works as described on the tutorial. Thanks for making the changes!