gopro/OpenGoPro #212: Camera Synchronization problem

pavlosharhan2, 2022-07-21:
I want to sync the footage from multiple GoPros for future video processing.

For this I would like to be able to get the video creation time with precision up to milliseconds. I did various attempts trying to get this info from MP4 metadata via ffprobe, but the timestamps there seem to have to sense.

My current solution is to get the Time from My PC while sending the "shutter on" command and take away ~0.132 seconds considering average latency between sending the command and actual start of the video. The error with this approach is ~20-40 ms, which is not enough for me. One more downside of this method is that latency is different for different presets. That's why I want something neater and more accurate.

The only one solution in my mind could be enabling GPS and extract the timestampes (with milliseconds precision) from telemetry files.
I feel like It's an overkill. It would drain the battery significantly and there might be no signal. I'd made some tests in past and sometimes the GoPro needs 5-10 minutes to catch the GPS signal.

Is there some better solution for this?

@tcamise-gpsw 
@KonradIT

KonradIT, 2022-07-21:
You need GoPro Labs.

https://gopro.github.io/labs/control/precisiontime/

https://gopro.github.io/labs/control/gpssync/

steelbytes, 2022-08-21:
syncronisation is important.  I also really need this via bluetoothle, not qr code

tcamise-gpsw, 2022-09-15:
What level of synchronization accuracy are you trying to achieve? You mentioned above that 20 ms is not enough.

steelbytes, 2022-09-18:
> What level of synchronization accuracy are you trying to achieve? You mentioned above that 20 ms is not enough.

I'm happy with say 100ms but others may(do) want better.

I want it so I can sync the video with telemetry data.  I am planning on writing an app that decodes the video, renders a speedo&tacho and reencodes the video.  my vehicle data is at a rate of 100ms.

Landstein, 2022-10-15:
@steelbytes  working on something similar.  Attempting to get two gopro devices to film simultaneously for a redundant recording.  if you'd be interested in collaboration or sharing ideas, get in touch.